The-OpenROAD-Project/OpenROAD #5636: Rsz degrades setup rather than improve it

maliberty, 2024-08-22:
### Describe the bug

I tried changing
```
--- a/flow/designs/asap7/swerv_wrapper/constraint.sdc
+++ b/flow/designs/asap7/swerv_wrapper/constraint.sdc
@@ -2,7 +2,7 @@ current_design swerv_wrapper
 
 set clk_name core_clock 
 set clk_port_name clk
-set clk_period 2500 
+set clk_period 1000
```

I observe in 4_1_cts.log that repair_timing makes things worse:
```
[INFO RSZ-0099] Repairing 930 out of 930 (100.00%) violating endpoints...
   Iter   | Removed | Resized | Inserted | Cloned |  Pin  |    WNS   |   TNS      |  Viol  | Worst
          | Buffers |  Gates  | Buffers  |  Gates | Swaps |          |            | Endpts | Endpt
---------------------------------------------------------------------------------------------------
        0 |       0 |       0 |        0 |      0 |     0 | -150.852 |  -102009.4 |    930 | swerv/dec/instbuff/bp0ff.genblock_dff.dout\[29\]$_DFF_PN0_/D
...
     2200 |       0 |       0 |      153 |      0 |    11 | -636.199 |  -636317.9 |    930 | mem/Gen_dccm_enable_dccm/mem_bank_1__dccm_bank/mem/addr_in[5]
    final |       0 |       0 |      153 |      0 |    11 | -636.199 |  -636317.9 |   1126 | mem/Gen_dccm_enable_dccm/mem_bank_1__dccm_bank/mem/addr_in[5]
```

### Expected Behavior

Improve timing (or at least neutral)

### Environment

```shell
With ORFS dadde560afdc443d2fb26c81ad4b8f628d398dff and OR 42ba30d730851a73bd2996c5673a05fd79520f87
```

### To Reproduce

as above

maliberty, 2024-08-22:
Note at period=1500 it still hits 0 tns and this problem is not observed.

precisionmoon, 2024-08-22:
This may benefit from odb based save/restore.   I will look into it.

precisionmoon, 2024-08-23:
There are multiple issues here.   One of them is RSZ journal save/restore.   Since this doesn't preserve the exact order of all the transforms, it's hard to do 100% restoration to a previous snapshot.   Another issue is the delay estimation during buffering.  It fails to predict worse QoR with additional buffering.  With SKIP_PIN_SWAP=1, big QoR degradation is mitigated.

```
repair_timing -verbose -repair_tns 100 -skip_pin_swap
[INFO RSZ-0094] Found 930 endpoints with setup violations.
[INFO RSZ-0099] Repairing 930 out of 930 (100.00%) violating endpoints...
   Iter   | Removed | Resized | Inserted | Cloned |  Pin  |    WNS   |   TNS      |  Viol  | Worst
          | Buffers |  Gates  | Buffers  |  Gates | Swaps |          |            | Endpts | Endpt
---------------------------------------------------------------------------------------------------
        0 |       0 |       0 |        0 |      0 |     0 | -150.852 |  -102009.4 |    930 | swerv/dec/instbuff/bp0ff.genblock_dff.dout\[29\]$_DFF_PN0_/D
       10 |       0 |       0 |       16 |      0 |     0 | -135.524 |   -88766.1 |    930 | swerv/dec/instbuff/bp0ff.genblock_dff.dout\[29\]$_DFF_PN0_/D
       20 |       0 |       0 |       26 |      0 |     0 | -135.524 |   -88766.1 |    930 | swerv/dec/instbuff/bp0ff.genblock_dff.dout\[29\]$_DFF_PN0_/D
       30 |       0 |       0 |       36 |      0 |     0 | -135.576 |   -88809.1 |    930 | swerv/dec/instbuff/bp0ff.genblock_dff.dout\[29\]$_DFF_PN0_/D

...
     1190 |       0 |       0 |      125 |      0 |     0 | -342.369 |  -319908.9 |    930 | mem/Gen_dccm_enable_dccm/mem_bank_1__dccm_bank/mem/addr_in[5]
     1193 |       0 |       0 |       79 |      0 |     0 | -134.332 |  -100334.9 |    930 | swerv/dec/instbuff/bp0ff.genblock_dff.dout\[29\]$_DFF_PN0_/D
     1200 |       0 |       0 |       85 |      0 |     0 | -134.332 |  -100334.9 |    930 | swerv/dec/instbuff/bp0ff.genblock_dff.dout\[29\]$_DFF_PN0_/D
     1200 |       0 |       0 |       85 |      0 |     0 | -134.332 |  -100334.9 |    930 | swerv/dec/instbuff/bp0ff.genblock_dff.dout\[29\]$_DFF_PN0_/D
    final |       0 |       0 |       85 |      0 |     0 | -134.332 |  -100334.9 |    989 | swerv/dec/instbuff/bp0ff.genblock_dff.dout\[29\]$_DFF_PN0_/D
---------------------------------------------------------------------------------------------------
```

A proper fix will need a project.
1) migrate from RSZ journal to odb save/restore
2) improve buffering (delay estimation and possibly sizing)
3) add codes to guarantee equal or better final QoR

povik, 2025-06-25:
Since this thread was active we have moved to odb save/restore and fixed a case of repair_timing not restoring to previous best result. @maliberty I think we can close it.