The-OpenROAD-Project/OpenROAD #5639: Google Test broke compiling on MacOS

vvbandeira, 2024-08-23:
This comit broke compling on MacOS.
"brew install googletest" is not suficcient, linking steps fail with
```
[ 73%] Built target odb_test_helper
ld: warning: ignoring duplicate libraries: '../../src/db/libdb.a', '../../src/zutil/libzutil.a'
ld: library 'gtest' not found
```

_Originally posted by @stefanottili in https://github.com/The-OpenROAD-Project/OpenROAD/issues/5630#issuecomment-2305918314_

vvbandeira, 2024-08-23:
@stefanottili, could you try to add the following to the dependency installer and re-run the script?
Looks like the version that Brew is installing by default is 1.15, and we have pinned in the common the 1.13
```
_installHomebrewPackage "googletest" "c3df14c9d6e05ec81029c0b1e2f3c3b9a7bd237b"
```

stefanottili, 2024-08-23:
a) zsh: command not found: _installHomebrewPackage

Where is this command supposed to be coming from ?
Why do you think that 1.15 won't work compared to 1.13 ?

b) make VERBOSE=1 shows that there is not path to /opt/homebrew/opt/googletest/lib in the link command
```
[ 94%] Linking CXX executable OdbGTests
cd /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build/src/odb/test/cpp && /opt/homebrew/Cellar/cmake/3.30.2/bin/cmake -E cmake_link_script CMakeFiles/OdbGTests.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++ -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/OdbGTests.dir/TestDbWire.cc.o CMakeFiles/OdbGTests.dir/TestAbstractLef.cc.o -o OdbGTests  -Wl,-rpath,/opt/homebrew/lib -lgtest -lgmock -lgtest_main ../../src/cdl/libcdl.a ../../src/defin/libdefin.a ../../src/def/libdef.a ../../src/def/libdefzlib.a ../../src/defout/libdefout.a ../../src/lefin/liblefin.a ../../src/lef/liblef.a ../../src/lef/liblefzlib.a /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/lib/libz.tbd ../../src/lefout/liblefout.a ../../src/db/libdb.a ../../src/zutil/libzutil.a ../../src/db/libdb.a ../../src/zutil/libzutil.a ../../../utl/libutl_lib.a /opt/homebrew/lib/libspdlog.1.9.2.dylib /opt/homebrew/lib/libfmt.8.1.1.dylib /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
ld: warning: ignoring duplicate libraries: '../../src/db/libdb.a', '../../src/zutil/libzutil.a'
ld: library 'gtest' not found
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/odb/test/cpp/OdbGTests] Error 1
make[1]: *** [src/odb/test/cpp/CMakeFiles/OdbGTests.dir/all] Error 2
make: *** [all] Error 2
```

stefanottili, 2024-08-23:
```
tools/OpenROAD/build/src/odb/test/cpp/CMakeFiles/OdbGTests.dir/link.txt
```
changing ```-lgtest -lgtest_main -lgmock``` to
```
/opt/homebrew/opt/googletest/lib/libgtest.a /opt/homebrew/opt/googletest/lib/libgmock.a /opt/homebrew/opt/googletest/lib/libgtest_main.a
```
allows me to build openroad.

I don't have enough cmake knowledge to get the static libs out of the google_tools cmake config files into link.txt

QuantamHD, 2024-08-24:
The brew package for googletest is missing a bunch of stuff. We really shouldn't rely on it.

stefanottili, 2024-08-25:
well, for the time being, this allows me to compile openroad:
change -lgtest -lgmock -lgtest_main in src/odb/test/cpp/CMakeFiles/OdbGTests.dir/link.txt to
/opt/homebrew/lib/libgtest.a /opt/homebrew/lib/libgmock.a /opt/homebrew/lib/libgtest_main.a

stefanottili, 2024-09-05:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/5692
fixed the link issue.

closed 2024-09-05 by maliberty