The-OpenROAD-Project/OpenROAD #5670: "VIRTUAL" keyword in net's routing is unsupported

bingyuew, 2024-08-29:
### Description

Got an error while reading a def file with the 'VIRTUAL' routing segments. The following is the log.

OpenROAD 016019ba7e265af5a7de05b9945d8a5b20654118
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 13177, timing group from output port.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 13210, timing group from output port.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 13243, timing group from output port.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 13276, timing group from output port.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 13309, timing group from output port.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 13342, timing group from output port.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 13375, timing group from output port.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 14771, timing group from output port.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 14804, timing group from output port.
[WARNING STA-1212] ./ASAP7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib line 14837, timing group from output port.
[INFO ODB-0227] LEF file: ./ASAP7/lef/asap7_tech_1x_201209.lef, created 24 layers, 9 vias
[WARNING ODB-0176] error: undefined layer (OVERLAP) referenced
[INFO ODB-0227] LEF file: ./ASAP7/lef/sram_asap7_64x256_1rw.lef, created 1 library cells
[INFO ODB-0227] LEF file: ./ASAP7/lef/asap7sc7p5t_27_R_1x_201211.lef, created 212 library cells
[WARNING ODB-0176] error: undefined layer (OVERLAP) referenced
[INFO ODB-0227] LEF file: ./ASAP7/lef/sram_asap7_16x256_1rw.lef, created 1 library cells
[WARNING ODB-0217] duplicate VIARULE (Pad_M9) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M9_M8) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M8_M7) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M7_M6) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M6_M5) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M3_M2widePWR0p936) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M4_M3widePWR0p864) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M5_M4widePWR0p864) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M6_M5widePWR1p152) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M7_M6widePWR1p152) ignoring...
[WARNING ODB-0217] duplicate VIARULE (M2_M1) ignoring...
[INFO ODB-0227] LEF file: ./ASAP7/lef/asap7_tech_1x_201209.lef
[WARNING ODB-0176] error: undefined layer (OVERLAP) referenced
[INFO ODB-0227] LEF file: ./ASAP7/lef/sram_asap7_64x64_1rw.lef, created 1 library cells
[INFO ODB-0127] Reading DEF file: ac97_top.def
[INFO ODB-0128] Design: ac97_top
[WARNING ODB-0126] error: VIRTUAL in net's routing is unsupported
ERROR (DEFPARS-6010): An error has been reported in callback. Error in file ac97_top.def at line 23959, on token ;.
Update the def file before parsing the file again.
[ERROR ODB-0421] DEF parser returns an error!
Error: test.tcl, 23 ODB-0421
openroad> exit

The attached file is the testcase. After decompressing it, please run:
1) cd testcase
2) <openroad binary> test.tcl
[testcase.tar.gz](https://github.com/user-attachments/files/16801701/testcase.tar.gz)

maliberty, 2024-08-29:
OR never generates such so they can only come from other tools.  You can use -continue_on_errors to continue.  What is the use case here?

bingyuew, 2024-08-30:
Thanks for the reply! We are trying to read a global routed design or a design with virtual clock net segments into OpenROAD and do further simulation. @liangrj2014 can comment on this.

maliberty, 2024-08-31:
What do you mean by 'simulation'?