The-OpenROAD-Project/OpenROAD #5687: build: duplicate message ids not detected

titan73, 2024-09-04:
### Describe the bug

Duplicate message ids not detected.
The script find_message.py is not called to do the check. It seems some build dependencies are missing.

### Expected Behavior

Error: ODB 0250 used 2 times, next free message id is 251
make[2]: *** [src/odb/src/db/CMakeFiles/db.dir/build.make:2611: src/odb/src/db/libdb.a] Error 1
make[2]: *** Deleting file 'src/odb/src/db/libdb.a'
make[1]: *** [CMakeFiles/Makefile2:2520: src/odb/src/db/CMakeFiles/db.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

### Environment

```shell
Latest git.
```

### To Reproduce

Get latest git and checkout commit-id I used to test and build openroad:
```
get fetch
git checkout d0bfb45bfe12c9c5a2a5f01a0d5e914f91885ad2
cd build
cmake ..
make
```

Then change the following code:
```
  if (_mode != defin::DEFAULT) {
    if (chip == nullptr) {
      _logger->error(utl::ODB, 250, "Chip does not exist");
    }
  } else if (chip != nullptr) {
    **_logger->error(utl::ODB, 251, "Chip already exists");**
  } else {
    chip = dbChip::create(_db);
  }
```

So the line in bold log a message with the same ids as 3 lines above (250) then rebuild:
```
sed -i 's/251/250/' ../src/odb/src/defin/definReader.cpp
make
```

No duplicate ids error is triggered, to force that do the following:
```
sed -i 's/251/250/' ../src/odb/src/defin/definReader.cpp
make
```

The error is triggered.

titan73, 2024-09-05:
Is anyone able to reproduce the issue?

maliberty, 2024-09-05:
The problem is that the message scanning is only depending on src/db and not the other subdirs:
```
messages(
  TARGET db
  OUTPUT_DIR ../..
  SOURCE_DIR ../..
)
```
messages.cmake needs to be modified to take a set of targets and odb updated to pass a set.

maliberty, 2024-09-05:
I think with 3.29 we could use odb as the target:

New in version 3.29: The <target> may be an [ALIAS target](https://cmake.org/cmake/help/latest/manual/cmake-buildsystem.7.html#alias-targets).

If you want to try installing and testing that would be great.

titan73, 2024-09-06:
Ok thanks I'll take a look.

titan73, 2024-09-07:
I installed cmake 3.29.0 and I tried changing the db makefile with:
```
messages(
  TARGET odb
  OUTPUT_DIR ../..
  SOURCE_DIR ../..
)
```
I get the same error as with 3.24.2.

When reading the cmake doc, I understand aliases are just just another way to reference a library.
So with add_library(MyLibraryAlias ALIAS MyLibrary), we can reference the library using both name MyLibraryAlias and MyLibrary.
It's not clear how this could help here.

titan73, 2024-09-07:
I tried to check messages in odb makefile instead of db's one. Since it's an INTERFACE we can't use add_custom_command with odb as target and POST_BUILD.
So I use add_custom_target:
```
add_custom_target(
  odb_messages ${CMAKE_SOURCE_DIR}/etc/find_messages.py > messages.txt
  WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
  DEPENDS
    db
    cdl
    defin
    defout
    lefin
    lefout
    zutil
    gdsin
)

add_dependencies(odb odb_messages)
```
This does the job. I tested extensively and didn't find any issue.
It is also more straightforward to do this check in odb.

maliberty, 2024-09-11:
Fixed by #5704

closed 2024-09-11 by maliberty