The-OpenROAD-Project/OpenROAD #5699: ERROR: cannot install on macOS if you are root or using sudo (not recommended for brew).

LagrangePoint3, 2024-09-05:
### Describe the bug

I tried to download the OpenROAD from sources locally. I'm using Mac with M1 chip.

In the procedure of clone and install dependencies by running the setup.sh script, as following,
sudo ./setup.sh

I got an error saying that: 

ERROR: cannot install on macOS if you are root or using sudo  (not recommended for brew).

Is there any other Mac users also facing the same problem? How do you fix this problem?

### Expected Behavior

No response

### Environment

```shell
ERROR: cannot install on macOS if you are root or using sudo  (not recommended for brew).
```

### To Reproduce

No response

### Screenshots

[image]

maliberty, 2024-09-05:
@stefanottili are things working for you now?

stefanottili, 2024-09-08:
I run the etc/DependencyInstaller which takes care of homebrew and whet and all compiles/builds