The-OpenROAD-Project/OpenROAD #5712: DRT: resole conflicts between frInst::getTransform and frInst::getUpdatedXform

osamahammad21, 2024-09-09:
### Description

The code in DRT keeps fluctuating between the use of getTransform and getUpdatedXform. getTransform cannot be applied directly to the master shapes as its offset is not the correct transform offset. In my opinion, getTransform is always wrong and useless, however some parts of the code get away with using it.

titan73, 2024-09-09:
So the idea is to convert the code to always use getUpdatedXform and remove getTransform at the end?

maliberty, 2024-09-09:
What is the goal of getUpdatedXform?  I'm used to seeing people use getTransform across hierarchies.  Why do you think it is wrong?

osamahammad21, 2024-09-10:
getUpdatedXform returns the correct transform that should be applied to the master shapes. However, getTransform returns a wrong one, because its origin point is not the actual origin point but the lower left point of the instance. I am still managing to understand how it is used correctly.

maliberty, 2024-09-10:
Is the transform different from what you would get in DEF for the instance?

osamahammad21, 2024-09-10:
yes, thus the need for getUpdateXform.

maliberty, 2024-09-10:
getUpdatedXform is pretty ugly.  Can we just use the normal transform and avoid all this?  I'm concerned about how much of an impact it might have but haven't scoped it myself.

closed 2025-02-11 by osamahammad21