The-OpenROAD-Project/OpenROAD #5713: GRT: refactor maze and maze3D

eder-matheus, 2024-09-09:
### Description

In FastRouteCore, the two main files that deal with the routing congestion are very long and could have a better readability. There are a few functions that needs special attention: `mazeRouteMSMDOrder3D`, `updateCongestionHistory`,  `mazeRouteMSMD`, `updateRouteType13D`, `updateRouteType23D`.

The goals of the refactoring are:

- [ ] Avoid repeated code.
- [ ] Remove dead code
- [ ] Break long functions into smaller ones
- [ ] Document the functions

maliberty, 2024-09-09:
Because of the C heritage of FR the grid is quite dumb and all the logic is in the router code.  It would be nice to have the grid support more high level operations which should also simplify the router somewhat.  For example adding/removing an edge that spans multiple gcells could be a single grid API call rather than a for-loop in the router code.

Divinesoumyadip, 2026-03-05:
Hi @eder-matheus  and @jfgava ,

I am an applicant for the GSoC 2026 "Incremental Global Routing on CUGR" project. Since FastRoute is the reference implementation for the CUGR incremental mode, diving into maze.cpp and maze3D.cpp to clean up this routing logic seems like the perfect way for me to ramp up on the codebase.

@jfgava , are you still actively working on this refactor?

If this issue is currently stalled, I would love to take it over. I am very interested in @maliberty's suggestion to implement a higher-level grid API for adding/removing edges to simplify the router code. I have a strong background in C++ algorithmic design and would be happy to draft a PR for these functions.

Let me know!