The-OpenROAD-Project/OpenROAD #5719: [ERROR RSZ-0005] Run global_route before estimating parasitics for global routing.

oharboe, 2024-09-10:
### Describe the bug

asdf

### Expected Behavior

asdf

### Environment

```shell
asdf
```

### To Reproduce

asfd

oharboe, 2024-09-10:
> What is requested here?  The message seems clear enough.

It comes immediately after make do-grt, so how is grt not done?

oharboe, 2024-09-10:
I dont understand what the error message is telling me: I get it *after* I have run global route and when I try to view the results.

maliberty, 2024-09-10:
It looks like you have quit and restarted in-between (I see the OR banner).  Is that correct?

oharboe, 2024-09-10:
yes: i run global route, then i open the result in the GUI.

maliberty, 2024-09-10:
@QuantamHD FYI the same as your issue.

closed 2024-09-18 by eder-matheus