The-OpenROAD-Project/OpenROAD #5725: Improve incremental repair running times so that megaboom can be done in 24 hours

oharboe, 2024-09-10:
### Description

repair_timing breaks the 24 hour budget for megaboom, so repair_timing has been disabled.

Note that repair_timing is not the biggest problem in megaboom(the clock period is ca. 3000ps now and I believe repair timing is a small fraction of that...), so this is a forward looking feature request to when macro placement has been addressed.

Untar https://drive.google.com/file/d/14-UYL5iUD1GWlL10sYCOPIB0jOpVYDzd/view?usp=sharing

```
./run-me-BoomTile-asap7-base.sh
```

### Suggested Solution

Make repair_timing faster

oharboe, 2024-09-11:
after 8 hours or so, the stack trace is:

```
__cxa_throw (@__cxa_throw:3)
sta::DmpPi::evalDmpEqns() (.cold) (@sta::DmpPi::evalDmpEqns() (.cold):20)
sta::DmpAlg::findDriverParams(double) (@sta::DmpAlg::findDriverParams(double):543)
sta::DmpPi::gateDelaySlew(double&, double&) (@sta::DmpPi::gateDelaySlew(double&, double&):21)
sta::DmpCeffDelayCalc::gateDelay(sta::Pin const*, sta::TimingArc const*, float const&, float, sta::Parasitic const*, std::map<sta::Pin const*, unsigned long, sta::PinIdLess, std::allocator<std::pair<sta::Pin const* const, unsigned long>>> const&, sta::DcalcAnalysisPt const*) (@sta::DmpCeffDelayCalc::gateDelay(sta::Pin const*, sta::TimingArc const*, float const&, float, sta::Parasitic const*, std::map<sta::Pin const*, unsigned long, sta::PinIdLess, std::allocator<std::pair<sta::Pin const* const, unsigned long>>> const&, sta::DcalcAnalysisPt const*):92)
sta::GraphDelayCalc::findDriverArcDelays(sta::Vertex*, sta::MultiDrvrNet const*, sta::Edge*, sta::TimingArc const*, std::map<sta::Pin const*, unsigned long, sta::PinIdLess, std::allocator<std::pair<sta::Pin const* const, unsigned long>>>&, sta::DcalcAnalysisPt const*, sta::ArcDelayCalc*) (@sta::GraphDelayCalc::findDriverArcDelays(sta::Vertex*, sta::MultiDrvrNet const*, sta::Edge*, sta::TimingArc const*, std::map<sta::Pin const*, unsigned long, sta::PinIdLess, std::allocator<std::pair<sta::Pin const* const, unsigned long>>>&, sta::DcalcAnalysisPt const*, sta::ArcDelayCalc*):111)
sta::GraphDelayCalc::findDriverEdgeDelays(sta::Vertex*, sta::MultiDrvrNet const*, sta::Edge*, sta::ArcDelayCalc*, std::array<bool, 2ul>&) (@sta::GraphDelayCalc::findDriverEdgeDelays(sta::Vertex*, sta::MultiDrvrNet const*, sta::Edge*, sta::ArcDelayCalc*, std::array<bool, 2ul>&):71)
sta::GraphDelayCalc::findDriverDelays1(sta::Vertex*, sta::MultiDrvrNet*, sta::ArcDelayCalc*) (@sta::GraphDelayCalc::findDriverDelays1(sta::Vertex*, sta::MultiDrvrNet*, sta::ArcDelayCalc*):108)
sta::GraphDelayCalc::findDriverDelays(sta::Vertex*, sta::ArcDelayCalc*) (@sta::GraphDelayCalc::findDriverDelays(sta::Vertex*, sta::ArcDelayCalc*):31)
sta::FindVertexDelays::visit(sta::Vertex*) (@sta::FindVertexDelays::visit(sta::Vertex*):72)
sta::BfsIterator::visit(int, sta::VertexVisitor*) (@sta::BfsIterator::visit(int, sta::VertexVisitor*):72)
sta::GraphDelayCalc::findDelays(int) (@sta::GraphDelayCalc::findDelays(int):48)
sta::Sta::searchPreamble() (@sta::Sta::searchPreamble():25)
sta::Sta::findRequireds() (@sta::Sta::findRequireds():7)
rsz::RepairSetup::repairSetupLastGasp(rsz::OptoParams const&, int&) (@rsz::RepairSetup::repairSetupLastGasp(rsz::OptoParams const&, int&):437)
rsz::RepairSetup::repairSetup(float, double, int, bool, bool, bool, bool, bool) (@rsz::RepairSetup::repairSetup(float, double, int, bool, bool, bool, bool, bool):739)
_wrap_repair_setup (@_wrap_repair_setup:124)
TclNRRunCallbacks (@TclNRRunCallbacks:38)
___lldb_unnamed_symbol1507 (@___lldb_unnamed_symbol1507:348)
Tcl_EvalEx (@Tcl_EvalEx:11)
```

oharboe, 2024-09-11:
@jeffng-or @precisionmoon @maliberty In addition to improving performance, can we add some logic to abandon futile repair timing automatically?

WNS is -3812.590ps for 3000ps clock period. Isn't repair_timing an exercise in futility at that point?

oharboe, 2024-10-22:
Add an option to stop repairing at a certain threshold?

TIMING_REPAIR_UNTIL=-1000

For exploring placement & global route issues in megaboom, then running repair for 10 minutes until you get from -20000 to -1200 might be more than good enough to work on placement & global route issue?

Running until -1000ps might take many more hours... Eventually for a final run when all issues are resolved, it is probably is worth doing the last gasp repair timing...

Without timing repair, WNS=-20000. Even after a few iterations, a few minutes, it is enormously improved to -1300 or so.

[image]

maliberty, 2024-10-22:
SETUP_SLACK_MARGIN/HOLD_SLACK_MARGIN already do this.  They were intended for over-fixing but nothing prevents you from using them for under-fixing afaik.

oharboe, 2024-10-22:
Nice! I just need to learn how to use them then. Any examples?

maliberty, 2024-10-22:
microwatt though with a positive value

oharboe, 2024-10-22:
> microwatt though with a positive value

Would setting a clock period that gives us a positive slack quickly in timing repair have the same effect?

megaboom has a 1200ps clock period and gets to negative slack -1200 relatively quickly, so I should see timing repair complete quickly if I set clock period to 1200+1200 = 2400ps?

Typo... https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2497

oharboe, 2024-10-22:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2498

maliberty, 2024-10-22:
> Would setting a clock period that gives us a positive slack quickly in timing repair have the same effect?

Yes.

oharboe, 2024-10-22:
Moot, use SETUP_SLACK_MARGIN with negative value to terminate retiming early