The-OpenROAD-Project/OpenROAD #5729: upf wrong location of isolation with clamp value 1

kareefardi, 2024-09-11:
### Describe the bug

when using isolation with `-location self` and `-clamp_value 1` an inverter is inserted such that the clamped value is inverted:
```
    - y_iso ( spm_inst_0/output35_y_iso_y_iso_o_isolation A ) ( spm_inst_0/output35 X ) + USE SIGNAL ;
    - y_iso_o ( spm_inst_1/input34 A ) ( spm_inst_0/output35_y_iso_y_iso_o_isolation_inv_out Y ) + USE SIGNAL ;

```
however the inserted inverter is not in the same group and region as the isolation cell: 
```
GROUPS 1 ;
    - spm_inst_0_domain spm_inst_0/output35_y_iso_y_iso_o_isolation spm_inst_0/output35 spm_inst_0/input9
         spm_inst_0/input8 spm_inst_0/input7 spm_inst_0/input6 spm_inst_0/input5

```

### Expected Behavior

I believe both should be in the same location 

### Environment

```shell
Git commit: c31d31057541e2c20515e657d5b958b04d2c365e
kernel: Linux 6.10.8-3-liquorix-amd64
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-15595-gc31d31057
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
[... 106 lines cut ...]
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found OpenMP: TRUE (found version "4.5")  
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/lNumber of processor cores: 20
ocal/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")  
-- Found Eigen3: /usr/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.qXxjHEjjq6
```

### To Reproduce

[test-case.tar.gz](https://github.com/user-attachments/files/16963696/test-case.tar.gz)

maliberty, 2024-09-11:
Is the expectation that the isolation cell and the inverter should both be outside the domain being isolated?  I want to confirm as that's what makes sense to me.

kareefardi, 2024-09-16:
> Is the expectation that the isolation cell and the inverter should both be outside the domain being isolated? I want to confirm as that's what makes sense to me.

To me what makes sense is that the both should be together based on the value of `-location`

maliberty, 2024-10-03:
I don't see how it makes sense to put the isolation cell in the domain it is isolating.  Unless it is connected to an always on rail it will just be turned off with the rest of the domain and provide no isolation.  

[OR doesn't support always-on rails]

maliberty, 2024-10-04:
I've added code to both put the inverter(s) in the right group but also get the sense default right so that no inverter is generated in the right place.

PR #5873 (2024-10-05): upf: set the group for isolation cell related inverters to the same g…
…roup

Also fix the defaults for isolation_sense (high) and location (self)

Fixes #5729
commit 94c274b6 (2024-10-05): upf: set the group for isolation cell related inverters to the same group

Also fix the defaults for isolation_sense (high) and location (self)

Fixes #5729
closed 2024-10-05 by maliberty