The-OpenROAD-Project/OpenROAD #5731: Improve `[ERROR MPL-0034] Specified location results in illegal placement. Cannot place macro outside of the core.`

oharboe, 2024-09-11:
### Description

untar  https://drive.google.com/file/d/11sRN_Br8SXQBMT1F8_yvJwwnZ8kGNpXM/view?usp=sharing

```
$ ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-15583-g45dd89bd8 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
core.FpPipeline.fregfile.regfile_ext
[ERROR MPL-0034] Specified location results in illegal placement. Cannot place macro outside of the core.
Error: macros.tcl, 2 MPL-0034
openroad> 
```

### Suggested Solution

add coordinates and sizes to error message so that it is clear what exactly is wrong.

oharboe, 2024-09-11:
@maliberty @AcKoucher There's a bug somewhere reproducible with the above test-case, problems with units...

I tried adding a [better error message](https://github.com/The-OpenROAD-Project/OpenROAD/pull/5735).

```
[INFO ORD-0030] Using 16 thread(s).
core.FpPipeline.fregfile.regfile_ext
[ERROR MPL-0034] Cannot place core.FpPipeline.fregfile.regfile_ext at ( 888598 1741440 ) ( 1186966 2140728 ), outside of the core ( 1026 1080 ) ( 2024190 2024190 ).
Error: macros.tcl, 2 MPL-0034

```

maliberty, 2024-09-11:
The rect is in DBU not microns.

maliberty, 2024-09-11:
See ClusteringEngine::init() for example:
```
  logger_->report(
      "Floorplan Outline: ({}, {}) ({}, {}),  Core Outline: ({}, {}) ({}, {})",
      block_->dbuToMicrons(die.xMin()),
      block_->dbuToMicrons(die.yMin()),
      block_->dbuToMicrons(die.xMax()),
      block_->dbuToMicrons(die.yMax()),
...
```

oharboe, 2024-09-11:
@maliberty :facepalm: 

Much better...

```
[ERROR MPL-0034] Cannot place core.FpPipeline.fregfile.regfile_ext at (888.598, 1741.44) (1186.966, 2140.728), outside of the core (1.026, 1.08) (2024.19, 2024.19).
```

closed 2024-09-11 by oharboe

maliberty, 2024-09-11:
FYI you can use 'fixes #<id>' in the commit message to close on merge, see https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue