The-OpenROAD-Project/OpenROAD #5733: Swapped pins on non-swappable instances

AcKoucher, 2024-09-11:
### Describe the bug

According to @precisionmoon, on sky130hd/gcd, by manually tracing eqy failure points, it's possible to notice swapped pins on some non-swappable instances like AOI. Disabling pin swapping (SKIP_PIN_SWAP=1 in  ORFS) results in eqy passing and flow completion.

### Expected Behavior

Eqy must not fail and non-swappable instance shouldn't have their pins swapped.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.0-1024-gcp
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-15498-gdcf73e99f
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
[... 111 lines cut ...]
-- Generating done
-- Build files have been written to: /tmp/tmp.x8tqICJmTo
```

### To Reproduce

Use the CI artifact in https://drive.google.com/file/d/1XDuesY9v2uendqtPyEZ8Dn7ik6_U698m/view?usp=sharing

### Relevant log output

```shell
Last log file 4_equivalence_check.log
Found 1 errors in the logs.
[ERROR CI-0000] No error message found. Here are the last 10 lines of the log.
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] Failed to prove equivalence of partition gcd.dpath.a_reg.out.4_DFFE_PP_.base.dff0.D
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] Failed to prove equivalence of partition gcd.dpath.a_reg.out.3_DFFE_PP_.base.dff0.D
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] Failed to prove equivalence of partition gcd.dpath.a_reg.out.15_DFFE_PP_.base.dff0.D
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] Failed to prove equivalence of partition gcd.dpath.a_reg.out.14_DFFE_PP_.base.dff0.D
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] Failed to prove equivalence of partition gcd.dpath.a_reg.out.12_DFFE_PP_.base.dff0.D
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] Failed to prove equivalence of partition gcd.dpath.a_reg.out.11_DFFE_PP_.base.dff0.D
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] Failed to prove equivalence of partition gcd.dpath.a_reg.out.10_DFFE_PP_.base.dff0.D
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:12 (12)
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] summary: Elapsed process time [H:MM:SS (secs)]: 0:01:12 (72)
EQY 15:14:02 [./logs/sky130hd/gcd/base/4_eqy_output] DONE (FAIL, rc=2)
```

AcKoucher, 2024-09-11:
As requested by @maliberty, @QuantamHD Could you please take a look at this?

QuantamHD, 2024-09-11:
I can take a look. The code seemed pretty bullet proof so I'm wondering how this happens. Do you know what standard cell in particular is incorrect?

AcKoucher, 2024-09-11:
Thanks!

> Do you know what standard cell in particular is incorrect?

 I didn't debug it. @precisionmoon Do you know about the incorrect std cell?

precisionmoon, 2024-09-11:
Eqy output is pretty cryptic but I saw a32o, a21oi cells in the transitive fanin of the failing outputs.

maliberty, 2024-09-14:
@QuantamHD any news?

QuantamHD, 2024-09-16:
I traced it back to this gate:

[image]

Yosys collapses the output of this gate to a constant which differs
[image]

Tracing all the way back to the first register suggests that the IO of this gate has been broken by something other than gate swap.

```
-- after resize --
net80 = \dpath.b_reg.out[2]$_DFFE_PP_
net81 = net80

-- before resize --
_037_ = \dpath.a_lt_b$in1[0]
\dpath.a_lt_b$in1[0] = \dpath.b_reg.out[0]$_DFFE_PP_
net3 = net2
net2 = \dpath.b_reg.out[2]$_DFFE_PP_
```

maliberty, 2024-09-16:
@QuantamHD how does this explain "Disabling pin swapping (SKIP_PIN_SWAP=1 in ORFS) results in eqy passing and flow completion."?

QuantamHD, 2024-09-16:
I think that's just random fluctuations leading to this thing not happening. I suspect if you disabled other resizer passes it may fix it as well.

I won't have time to look at this again for a few days, but I don't think turning off pin swap is going to solve this problem. There seems to something else happening to instance _455_ that's causing an issue here.

precisionmoon, 2024-09-16:
Is this a Yosys issue or an OR issue?   OR doesn't do any register collapsing.

maliberty, 2024-09-16:
Its an OR issue - the check is across the call to repair_timing

QuantamHD, 2024-09-16:
EQY is collapsing the circuit for formal equivalence, but in the before and after resize you can see from my analysis that the netlist isn't doing the same thing anymore.

precisionmoon, 2024-09-16:
Based on the analysis from @QuantamHD, net3 which was attached to C pin of and4b_1 became, after resizing, net81 which is attached to B pin of and4b_1.  It is possible that during B and C pin swapping, the connectivity got messed up, perhaps due to direct pin-to-pin wires without buffers.  Let me dig into this a bit more.

maliberty, 2024-09-16:
Let's do B as we keep hitting these and that's the best solution.

maliberty, 2024-09-16:
When the restoration fails, does it fail silently?

precisionmoon, 2024-09-16:
Yes, it doesn't know it failed.

maliberty, 2024-10-02:
@precisionmoon is this resolved now with the odb journal?

precisionmoon, 2024-10-02:
> @precisionmoon is this resolved now with the odb journal?

Yes.

closed 2024-10-02 by precisionmoon