The-OpenROAD-Project/OpenROAD #5750: No clock with fakeram

oharboe, 2024-09-16:
### Describe the bug

untar [fakeram-no-clock.tar.gz](https://github.com/user-attachments/files/17012789/fakeram-no-clock.tar.gz)

Run:

```
./run-me-top-asap7-fakeram.sh
```

```
OpenROAD v2.0-15655-gdc5f404d7
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0227] LEF file: /home/oyvind/bazel-orfs/foo/platforms/asap7/lef/asap7_tech_1x_201209.lef, created 24 layers, 9 vias
[INFO ODB-0227] LEF file: /home/oyvind/bazel-orfs/foo/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef, created 212 library cells
[INFO ODB-0227] LEF file: sram/fakeram/sdq_17x64.lef, created 1 library cells
The design is gutted when mocking floorplan
Clock                   Period          Waveform
----------------------------------------------------
openroad> 
```

### Expected Behavior

`clk` should be listed by `report_clock_properties`

### Environment

```shell
OpenROAD v2.0-15655-gdc5f404d7
```

### To Reproduce

See above

oharboe, 2024-09-16:
@maliberty @jeffng-or FYI

maliberty, 2024-09-16:
How was the ram generated?  fakeram2.0 always creates a clock pin.

oharboe, 2024-09-16:
> How was the ram generated?  fakeram2.0 always creates a clock pin.

@jeffng-or ?

oharboe, 2024-09-17:
@maliberty @jeffng-or My understanding:

Any macro with a sequential element, will be listed in all_registers at the next level up. This is not the case with the fakeram .lib and OpenROAD. If I create a macro from the fakeram.v file, then that macro is listed in all_registers the next level up.

So does this mean that the fakerame .lib file does not contain a sequential element or that OpenROAD can't detect it?

Since there is no sequential element detected, all_registers is empty and finally there's no clock. The root cause is that OpenSTA detects no sequential element.

gadfort, 2024-09-17:
@oharboe I believe it is any element that is a sequential is in the list, macros are different.

oharboe, 2024-09-17:
> @oharboe I believe it is any element that is a sequential is in the list, macros are different.

I know from other designs, such as mock-array, that any macro that contains a sequential element is listed in all_registers the next level up. So Element in mock-array is listed in mock-array with all_registers.

I know from other .lib files from real PDKs that an instantiated SRAM from a .lib/.lef file is listed in all_registers.

However, this, strangely is not the case for the fakeram .lib in this github issue.

jeffng-or, 2024-09-17:
> > How was the ram generated?  fakeram2.0 always creates a clock pin.
> 
> @jeffng-or ?

FakeRAM 2.0 always creates a clock pin. The Liberty was generated using one of the ASAP7 SRAMs as a template. It also creates a clock pin and sets the clock pin attribute to true:

    pin (clk) {
      clock : true;

maliberty, 2024-09-17:
```
1566a1567
>         timing_type : rising_edge;
1568d1568
<         timing_type : combinational;
```
fixes it.  @jeffng-or please look at what in fakeram needs changing.

maliberty, 2024-09-17:
Though I see nothing in fakeram2.0 that generates such an edge...?

oharboe, 2024-09-17:
@jeffng-or With the fix from @maliberty https://github.com/The-OpenROAD-Project/bazel-orfs/pull/159

closed 2024-09-17 by maliberty

jeffng-or, 2024-09-17:
> Though I see nothing in fakeram2.0 that generates such an edge...?

The Liberty files were not generated from FakeRAM, but from a script that I put together that uses the https://github.com/The-OpenROAD-Project/asap7_sram_0p0/blob/main/generated/LIB/srambank_128x4x16_6t122.lib as a template.

I'll take a look at updating the script and re-generating the Liberty files.