The-OpenROAD-Project/OpenROAD #5755: Tech vias with multiple cuts only generate a single cut in pdngen

gatecat, 2024-09-17:
### Describe the bug

I think this is a regression in https://github.com/The-OpenROAD-Project/OpenROAD/commit/ab7c0f0e5fca2f1a572b7b7337dc80988e1099c3, I believe this was working back in April.

When `add_pdn_connect` is forced to use a multiple-cut tech via using `fixed_vias`, only one cut per tech via instantiation is actually generated.

### Expected Behavior

There's a 3-cut tech file in our (made-up, but representative of an actual case) tech lef called `via1_0_x3h` that we force pdngen to use. In this case, the tech via is repeated 4 times horizontally. 12 cuts should be generated, but in fact, only 4 are generated:

[image]

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.10.10-arch1-1
os: Arch Linux 
cmake version 3.30.3
CMake Warning at CMakeLists.txt:106 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 14.2.1
```

### To Reproduce

As a shareable minimum example, I've created a version of nangate45 with a three-cut tech via called "via1_0_x3h". The entire example is uploaded here (run `nangate45_multicut.tcl`): 

[nangate45_multicut.zip](https://github.com/user-attachments/files/17028331/nangate45_multicut.zip)

gatecat, 2024-09-17:
I just double checked, and indeed reverting https://github.com/The-OpenROAD-Project/OpenROAD/commit/ab7c0f0e5fca2f1a572b7b7337dc80988e1099c3 makes this work as expected:
[image]

maliberty, 2024-09-17:
@gadfort thoughts?

gadfort, 2024-09-17:
@maliberty I'll take a look this afternoon, I do recall there being some odd behavior with multicut vias.

PR #5759 (2024-09-18): pdn: multicut handling
Closes #5755 

Changes:
- iterates over internal vias in multicut vias to fill in missing vias

No new test needed since it turns  out the asap7_offcenter_via test exhibited the same issue. 

Possible improvements:
- there is a possibility to further reduce the number of instances by collapsing the who via into a single instance, however, that is causing of odd side effects, so I think it would make sense to handle that in a separate PR.