The-OpenROAD-Project/OpenROAD #5769: dangling output of rebuffer4

maliberty, 2024-09-18:
### Describe the bug

I was looking at the drt failure in asap7/ethmac_lvt and noticed that rebuffer4/Y is connected to a net with no outputs.  I don't know why we would have such a case as rebuffering should require a load.

Also odd is that this net has a guide (@eder-matheus any ideas?).

### Expected Behavior

Either a load or no rebuffer4.

### Environment

```shell
https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Nightly-Public/4515/

ORFS: 74df8de6bf80af9197ed51285489a088b29af841
OR: b061a21c943fadaca8f972a73f802eca15449935
```

### To Reproduce

Use the artfacts from https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Nightly-Public/4515/artifact/flow/final_report_ethmac_lvt_asap7_base_2024-09-18_13-39.tar.gz

DESIGN_CONFIG=designs/asap7/ethmac_lvt/config.mk make gui_grt

inspect net1503

### Screenshots

[image]

[image]

eder-matheus, 2024-09-18:
@maliberty My guess is that this net had more pins in the beginning of GRT flow, and it created these guides for it. But later during incremental GRT, the net ends with only 1 pin, but GRT didn't updated the routing for it. I will take a look.

maliberty, 2024-09-18:
The BufferTree for the input of rebuffer4 is also odd - too many buffers for such a small area:
[image]

maliberty, 2024-09-18:
@osamahammad21 FYI on the underlying trigger for drt

precisionmoon, 2024-09-18:
I see 2 removed buffers and 45 swapped pin instances.

[INFO RSZ-0059] Removed 2 buffers.
[INFO RSZ-0040] Inserted 18 buffers.
[INFO RSZ-0041] Resized 19 instances.
[INFO RSZ-0043] Swapped pins on 45 instances.
[INFO RSZ-0049] Cloned 3 instances.

This may be the same RSZ journal restore bug of issue [5733](https://github.com/The-OpenROAD-Project/OpenROAD/issues/5733).   The issue may go away with SKIP_PIN_SWAPPING=1.