The-OpenROAD-Project/OpenROAD #5818: drt error when port is located above max routing layer

gadfort, 2024-09-27:
### Describe the bug

When then max routing layer is set below the the pin locations for some ports, drt fails with:
```
[ERROR DRT-0255] Maze Route cannot find path of net clk_i in worker of routeBox ( 0 7560 ) ( 3780 11340 ).
```

### Expected Behavior

- GRT should have generated an error since this port could not be reached within the routing layers provided.
- DRT should have generated an error much earlier to that this port cannot be access, the `[ERROR DRT-0255] Maze Route cannot find path of net clk_i in worker of routeBox ( 0 7560 ) ( 3780 11340 ).` is pretty obscure for a user to track down.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.0-45-generic
os: Ubuntu 24.04.1 LTS (Noble Numbat)
cmake version 3.24.2
-- The CXX compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16075-g57a1b4c0c
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 132 lines cut ...]
[INFO DRT-0179] Init gr pin query.
[INFO DRT-0185] Post process initialize RPin region query.
[INFO DRT-0181] Start track assignment.
[INFO DRT-0184] Done with 61494 vertical wires in 3 frboxes and 44080 horizontal wires in 3 frboxes.
[INFO DRT-0186] Done with 6277 vertical wires in 3 frboxes and 4957 horizontal wires in 3 frboxes.
[INFO DRT-0182] Complete track assignment.
[INFO DRT-0267] cpu time = 00:00:04, elapsed time = 00:00:02, memory = 1058.83 (MB), peak = 1058.83 (MB)
[INFO DRT-0187] Start routing data preparation.
[INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 1058.83 (MB), peak = 1058.83 (MB)
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
Failed to find a path between pin clk_i and source aps:
( 292 409 5 ) (Idx) / ( 3780 10064 ) (coords)
extBBox (xDim, yDim, zDim) = (449, 710, 10)
[ERROR DRT-0255] Maze Route cannot find path of net clk_i in worker of routeBox ( 0 7560 ) ( 3780 11340 ).
    Completing 10% with 0 violations.
    elapsed time = 00:00:00, memory = 3956.00 (MB).
    Completing 20% with 0 violations.
    elapsed time = 00:00:01, memory = 2832.50 (MB).
Error: sc_route.tcl, 126 DRT-0255
```

### Additional Context

I think two fixes are needed, 1 in GRT and 1 in DRT.

maliberty, 2024-09-28:
I believe the intention is to allow the router to go above the max routing layer when the pin (bump) is up there but you don't want to use those layers for general routing.  Therefore just a drt issue.

osamahammad21, 2024-10-08:
I am getting this error:
[ERROR ORD-0054] odb file inputs/ibex_core.odb is invalid: basic_ios::clear: iostream error
Probably something went wrong with backward compatibility, Could you send me the OR commit you used?

gadfort, 2024-10-08:
@osamahammad21 -- OpenROAD version: v2.0-16075-g57a1b4c0c

osamahammad21, 2024-10-09:
We usually add a stack of vias from the Pin down to the TOP_ROUTING_LAYER. The problem in this case is the position of that pin. 
[image]
There is no routing track passing through it. This makes it hard for this case where RIGHTWAYONGRIDONLY is enforced as that makes the only option to reach the accesspoint is through a via. But the lower layer doesn't have a track on that position as well. This may suggest that we are not handling this case of pin access correctly in DRT but for now a simple solution is to choose a better location for the via stack. Now it routes. I'll close the issue on the PR merge and follow up with other PA and DR issues for this corner case.

maliberty, 2024-10-09:
@gadfort is it intended for the pin to be off-track?

gadfort, 2024-10-09:
@maliberty it's placed with `place_pin` so I suppose it's possible that it placed off-track.

osamahammad21, 2024-10-09:
The pin itself is not offtrack, but the via on the top routing layer added by the detailed router to access the pin is.

closed 2024-10-23 by osamahammad21