The-OpenROAD-Project/OpenROAD #5829: Stack Dump

ashkanr65, 2024-09-30:
### Describe the bug

"report_power -corner $power_corner" line  returns this error:
report_power -corner $power_corner
Signal 11 received
Stack trace:
 0# 0x000055E3931560EB in openroad
 1# 0x00007FCCB1EB6520 in /lib/x86_64-linux-gnu/libc.so.6
 2# sta::Network::findPin(sta::Instance const*, sta::LibertyPort const*) const in openroad
 3# sta::Power::seedRegOutputActivities(sta::Instance const*, sta::Sequential*, sta::LibertyPort*, bool) in openroad
 4# sta::Power::seedRegOutputActivities(sta::Instance const*, sta::BfsFwdIterator&) in openroad
 5# sta::Power::ensureActivities() in openroad
 6# sta::Power::power(sta::Corner const*, sta::PowerResult&, sta::PowerResult&, sta::PowerResult&, sta::PowerResult&, sta::PowerResult&, sta::PowerResult&) in openroad
 7# design_power(sta::Corner const*) in openroad
 8# 0x000055E3933A33BD in openroad
 9# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
10# 0x00007FCCB637D028 in /lib/x86_64-linux-gnu/libtcl8.6.so
11# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
12# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
13# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
14# ord::tclAppInit(Tcl_Interp*) in openroad
15# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
16# main in openroad
17# 0x00007FCCB1E9DD90 in /lib/x86_64-linux-gnu/libc.so.6
18# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
19# _start in openroad
Segmentation fault

### Expected Behavior

report_power -corner $power_corner

### Environment

```shell
Git commit: caa0853832b0605e7d26f85faf9dc918675e4636
kernel: Linux 5.15.153.1-microsoft-standard-WSL2
os: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16120-gcaa085383
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
[... 117 lines cut ...]
-- Found OpenMP: TRUE (found version "4.5")  
-- Found OpenMP: TRUE (found version "4.5")  
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/Number of processor cores: 16
cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.eHhYlyXaiE
```

### To Reproduce

I cannot upload the files but the error happened after this command on flow.tcl:
report_power -corner $power_corner

maliberty, 2024-09-30:
Without a test case I can do much.  My guess is some sort of nullptr access.  You'll have to debug a bit further to give some indication of the issue or engage a support contract with https://precisioninno.com/ to pass data under NDA.

ashkanr65, 2024-09-30:
> Without a test case I can do much. My guess is some sort of nullptr access. You'll have to debug a bit further to give some indication of the issue or engage a support contract with https://precisioninno.com/ to pass data under NDA.

If you drop your email, I can send it to you by email.

maliberty, 2024-09-30:
mliberty@precisioninno.com but I can't guarantee when I'll have time to look.

QuantamHD, 2024-09-30:
@ashkanr65 if you compile OpenROAD with -DASAN=ON it should give additional information about the memory error

maliberty, 2024-09-30:
Please don't send anything which would violation any NDAs you might have.

ashkanr65, 2024-10-01:
> [mliberty@precisioninno.com](mailto:mliberty@precisioninno.com) but I can't guarantee when I'll have time to look.

@maliberty I already sent you the case.
Thank you.

ashkanr65, 2024-10-01:
> @ashkanr65 if you compile OpenROAD with -DASAN=ON it should give additional information about the memory error

I will try that too.

ashkanr65, 2024-10-01:
@QuantamHD I don't know how to get the additional information. I already compile openroad with -DASAN=ON and got the error...

maliberty, 2024-10-01:
User problem resolved off-line.

closed 2024-10-01 by maliberty