The-OpenROAD-Project/OpenROAD #5831: pgpins with name ending with ! (inherit) are not supported in liberty files

titan73, 2024-09-30:
### Describe the bug

When reading .lib with e.g voltage_map(vdd!, 1.55);
The following error occurs:
[ERROR STA-0026] xxx.lib line 82, syntax error

### Expected Behavior

No error

### Environment

```shell
Latest version
```

### To Reproduce

Take whatever test case with modified .lib with ! on a pgpins.

titan73, 2024-09-30:
Note that lef parser does not complain.

titan73, 2024-09-30:
Ah. liberty parser seems to be in OpenSTA. Should I create a issue or move this one there?

maliberty, 2024-09-30:
Create an issue in https://github.com/parallaxsw/OpenSTA

maliberty, 2024-09-30:
Be sure to include a small test case

closed 2024-09-30 by maliberty

titan73, 2024-09-30:
> Create an issue in https://github.com/parallaxsw/OpenSTA

Ok. I'll do that.

So once it fixed there, how long will it take to get integrated in https://github.com/The-OpenROAD-Project/OpenSTA which OR uses?

Btw, how is it that https://github.com/The-OpenROAD-Project/OpenSTA is not a fork of the parallaxsw repo?

maliberty, 2024-09-30:
It varies, we tend to pull sta when needed.  You can always request an update.

Its not a fork for historical reasons.  Don't file issues there.