The-OpenROAD-Project/OpenROAD #5837: ASAP7 ibex_core reports timing through two output pins on the same cell

b62833, 2024-10-01:
### Describe the bug

When I run

make DESIGN_CONFIG=./designs/asap7/ibex/config.mk FLOW_VARIANT=maybe_timing_bug

on ORFS commit 

53873f6b5a049a999386861e7a9ee80aa61080c5 

the section 

"finish report_checks -path_delay max"

of the file

reports/asap7/ibex/maybe_timing_bug/6_finish.rpt

has these lines:

```
 447      1    1.22    9.79   18.86  640.51 v _24962_/Y (AO21x1_ASAP7_75t_R)
 448                                          _18148_ (net)
 449                   9.80    0.16  640.67 v _35146_/A (HAxp5_ASAP7_75t_R)
 450      5    4.14   52.65   27.89  668.56 ^ _35146_/CON (HAxp5_ASAP7_75t_R)
 451                                          _01184_ (net)
 452      1    0.72   18.26   15.63  684.20 v _35146_/SN (HAxp5_ASAP7_75t_R)
 453                                          _01150_ (net)
 454                  18.26    0.05  684.25 v _22681_/A (BUFx5_ASAP7_75t_R)
```

This shows the timing path going in the pin A of the cell `_35146_`, out the pin CON of the same cell and then _out again_ of the pin SN on the same cell. 

This isn't logically possible. And the NLDM timing models I looked at don't have any timing arcs between CON and SN. So the timer is reporting something incorrect here.

### Expected Behavior

OpenROAD should report timing correctly.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.0-45-generic
os: Ubuntu 24.04.1 LTS (Noble Numbat)
cmake version 3.24.2
-- The CXX compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16089-gd06039e7d
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
[... 127 lines cut ...]
-- Found OpenMP: TRUE (found version "4.5")  
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARNumber of processor cores: 16
IES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.t6iMSnugiH
```

### To Reproduce

make DESIGN_CONFIG=./designs/asap7/ibex/config.mk FLOW_VARIANT=maybe_timing_bug

with the commit above.

maliberty, 2024-10-01:
There are output to output arcs on the half adder if you look at the liberty.  Since they are stored in .gz I can't provide a link but they are clearly there.  Roughly
```
    pin (SN) {
      direction : output;
      timing () {
        related_pin : "CON";
```

closed 2024-10-01 by maliberty

b62833, 2024-10-01:
Doh! Didn't look down far enough.

An output to output pin timing arc still looks goofy. 

But after digging through the asap7sc7p5t_27 cdl, that looks legitimate. Seems like a really bad idea to have the timing of one output depend on the loading of another output.

maliberty, 2024-10-01:
I agree it's a bad idea but that's how it is