The-OpenROAD-Project/OpenROAD #5847: Slow annealing for pins when using multiple metal layers for pins

oharboe, 2024-10-02:
### Description

untar [io-place-slow-annealing.tar.gz](https://github.com/user-attachments/files/17228305/io-place-slow-annealing.tar.gz)

Finishes immediately(seconds):

```
./run-me-mock-array_Element-asap7-base.sh
```

Didn't wait for it to complete:

```
PLACE_PINS_ARGS=-annealing ./run-me-mock-array_Element-asap7-base.sh
```

### Suggested Solution

Faster annealing

oharboe, 2024-10-10:
@maliberty https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2412 takes at least six hours without finishing pin placement

eder-matheus, 2024-10-11:
> [@maliberty](https://github.com/maliberty) [The-OpenROAD-Project/OpenROAD-flow-scripts#2412](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2412) takes at least six hours without finishing pin placement

This issue is the next on my list. Hopefully we'll have the fix for it soon.

closed 2024-10-12 by maliberty