The-OpenROAD-Project/OpenROAD #5848: [ERROR PDN-0233] Failed to generate full power grid.

oharboe, 2024-10-02:
### Describe the bug

From https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2412

untar [pdn-error.tar.gz](https://github.com/user-attachments/files/17228394/pdn-error.tar.gz)

```
./run-me-mock-array-asap7-base.sh 
```

```
OpenROAD v2.0-16219-g56c3c86ba
[deleted]
[WARNING PDN-0232] ElementGrid - ces_0_0 does not contain any shapes or vias.
[ERROR PDN-0233] Failed to generate full power grid.
Error: pdn.tcl, 5 PDN-0233
```

### Expected Behavior

No errors

### Environment

```shell
OpenROAD v2.0-16219-g56c3c86ba
```

### To Reproduce

See above

maliberty, 2024-10-02:
Is "ElementGrid - ces_0_0 does not contain any shapes or vias." false?

oharboe, 2024-10-06:
> Is "ElementGrid - ces_0_0 does not contain any shapes or vias." false?

I don't know how to check this.

maliberty, 2024-10-06:
In master the element has obstruction up to M5 but in your case they go up to M7.  As the grid is on M6 it can't be connected to.   I see you changed the MAX_ROUTING_LAYER which might relate to the difference.  When building the Element the routing should not be above the top pdn layer.

oharboe, 2024-10-06:
How can I tell what the max PDN layer is?

It seems very restrictive that top routing layers can only be used at the top/chip level.

maliberty, 2024-10-06:
You look at https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/edcd265180fc6f2edc04b6743920fc027c56c797/flow/designs/asap7/mock-array/config.mk#L37 which tells you to see https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/edcd265180fc6f2edc04b6743920fc027c56c797/flow/platforms/asap7/openRoad/pdn/BLOCKS_grid_strategy.tcl#L24 where the max stripe is on M6.  You need to be able to connect to the macro from above in most cases.  You also may want to be able to route over the macro.  If not you need to change the PDN definition.

oharboe, 2024-10-07:
@maliberty Thanks! PR updated, closing this issue.

closed 2024-10-07 by oharboe