The-OpenROAD-Project/OpenROAD #5849: [ERROR DRT-0218] Guide is not connected to design.

kareefardi, 2024-10-02:
### Describe the bug

Following the suggestion here https://github.com/The-OpenROAD-Project/OpenROAD/discussions/5507#discussioncomment-10250593 on how to run `repair_antennas` after detailed routing, OpenROAD generates the error reported in the title. Note that this specifically happens when `repair_antennas -iterations 2` is set to higher than 1 iterations. 

### Expected Behavior

Unless there is something wrong with the script, this error shouldn't happen

### Environment

```shell
Git commit: 56c3c86ba5aea97d1a6c18e90c8d2076c5740929
kernel: Linux 6.10.12-1-liquorix-amd64
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16219-g56c3c86ba
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
[... 137 lines cut ...]
global_route -start_incremental -allow_congestion -verbose
repair_antennas -iterations 2
detailed_route -save_guide_updates -verbose 1 -bottom_routing_layer met1 -top_routing_layer met5
```

### Relevant log output

```shell
[INFO DRT-1000] Pin _06122_/X not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[INFO DRT-1000] Pin _06129_/A1 not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[INFO DRT-1000] Pin _06161_/B not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[INFO DRT-1000] Pin _06163_/X not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[INFO DRT-1000] Pin _06164_/X not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[INFO DRT-1000] Pin _06172_/C not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[INFO DRT-1000] Pin _06171_/X not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[INFO DRT-1000] Pin _06172_/X not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[INFO DRT-1000] Pin _06174_/X not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[WARNING DRT-1001] No guide in the pin neighborhood
[WARNING DRT-0215] Pin _06174_/X not covered by guide.
[WARNING DRT-0215] Pin _06174_/X not covered by guide.
[WARNING DRT-0225] _02593_ 2 pin not visited, fall back to feedthrough mode.
[WARNING DRT-0224] _02593_ 2 pin not visited, number of guides = 4.
[ERROR DRT-0218] Guide is not connected to design.
Error: run.tcl, 10 DRT-0218
```

eder-matheus, 2024-10-02:
For post-DRT repair_antennas, I don't recommend to use the -iterations keyword for this command, since in the second iteration the antennas will be evaluated following the route guides. I suggest something similar to what OR regression tests and ORFS do:
OR regression: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/test/flow.tcl#L244-L272
ORFS: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/detail_route.tcl#L53-L61

In any case, this error should not be happening. I will investigate what is happening here.

kareefardi, 2024-10-03:
> For post-DRT repair_antennas, I don't recommend to use the -iterations keyword for this command, since in the second iteration the antennas will be evaluated following the route guides. I suggest [...]

Thank you. In the sections that you posted, I noticed that `-iterations` is not used at all and the loop is capped at `5` iterations. Is there a specific reason for the cap?

eder-matheus, 2024-10-03:
@kareefardi I was able to complete your design with zero antenna violations doing the following updates to your script:

```
read_db ./picorv32.odb
read_liberty ./sky130_fd_sc_hd__tt_025C_1v80.lib
read_sdc ./base.sdc
set_thread_count 20
remove_fillers
set_global_routing_layer_adjustment * 0.4
set_routing_layers -signal met1-met5 -clock met1-met5

set repair_antennas_iters 1
while {[check_antennas] && $repair_antennas_iters < 5} {
  repair_antennas
  detailed_route -save_guide_updates -verbose 1 -bottom_routing_layer met1 -top_routing_layer met5
  incr repair_antennas_iters
}

check_antennas
```

It takes two iterations to reach zero violations.

closed 2024-10-03 by maliberty

kareefardi, 2024-10-07:
> @kareefardi I was able to complete your design with zero antenna violations doing the following updates to your script:
>
> ```
> read_db ./picorv32.odb
> read_liberty ./sky130_fd_sc_hd__tt_025C_1v80.lib [...]

@eder-matheus I noticed you are not using `global_route -start_incremental`. Is it not needed?

eder-matheus, 2024-10-07:
> @eder-matheus I noticed you are not using `global_route -start_incremental`. Is it not needed?

For post-DRT repair_antennas it is not needed. repair_antennas itself is responsible to store and call grt to reroute the modified nets. At some point, I would like to not need to call `global_route -start_incremental` for any command, but for now only repair_antennas has support for it.