The-OpenROAD-Project/OpenROAD #5867: DRT: incremental Pin Access with odb callbacks

eder-matheus, 2024-10-03:
### Description

During incremental grt, the APs stored in odb can be invalidated when moving instances or changing their orientation. DRT should be able to update the APs for the iterms from modified instances incrementally, using the odb callbacks to understand what should be updated.

maliberty, 2024-10-03:
I think this could be good for @bnmfw as he has been learning the PA code.

bnmfw, 2024-10-04:
I don't yet have the full picture for the PA module, but from what I understand right now there is no way to incrementally change access to instances. The starting point of the PA is `FlexPA::prep()` where every pin access is determined at once. Today me an @eder-matheus had a meeting where we saw that **apparently** during incremental DRT the whole PA flow is done from scratch.  

### How It could be done
Determining the Access Points of an instance could be done in a vacuum. `initAllAccessPoints()` could be refactor to have a function responsible determining the Access Points of an instance. (Right now it iterates over instances and then pins directly).

Determining the Access Patterns of an instance could **probably** be done in a vacuum. `prepPatternInst()` already determines Access Patterns for a single instance. I'm still refactoring this part of code so I'm not 100% sure if it would be safe to call this function alone.

The problem starts in the instance rows. I still haven't got to this part of code so I'm speculating, once I'm more firm about this I will update this comment. `genInstRowPattern()` is responsible for determining the used Access Patterns of a single row. Changing an instance and its Access Patterns might change the Access Patters used for the other instances in the row, therefore most probably major changes would have to be done to allow an instance to change in a row with minimal changes to other instances in the row.

bnmfw, 2025-04-04:
I'm using this message to mark the point where a prototype of incremental PA is working. It currently set to only work in bulk, which needs to be changed. The next few PRs will set the structures for incremental to work instance by instance.

PR #7082 (2025-06-05): drt: incremental PA
Fixes #5867.

ISPD and Secure CI safe.

Pin Access is now Alive! Once Pin Access has a initial batch run done all its structures are kept alive by TritonRoute. DesignCallBack has its callbacks modified to allow for the update of PA data of any instance called.
During premove all PA data is erased, during postmove all PA data is calculated, if the moved instance is a new unique the unique data will also be calculated. In any case its row of instanced will have their access patterns solved again.
There is still a missing callback for creating a new inst.

#### `DesignCallBack [... cut]
closed 2025-06-05 by osamahammad21