The-OpenROAD-Project/OpenROAD #5875: Mangled tcl environment

oharboe, 2024-10-04:
### Describe the bug

```
sudo apt install tcllib
```

```
oyvind@corona:~/OpenROAD-flow-scripts/flow$ tclsh
% package require yaml
0.4.1
%
```

Whereas:

```
$ openroad
OpenROAD v2.0-16219-g56c3c86ba 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> package require yaml
error reading package index file /usr/lib/blt2.5/pkgIndex.tcl: Error: pkgIndex.tcl, 59 can't read "dir": no such variable
error reading package index file /usr/lib/tcltk/x86_64-linux-gnu/tclreadline2.3.8/pkgIndex.tcl: Error: pkgIndex.tcl, 11 can't read "dir": no such variable
[etc]
can't find package yaml
while evaluating package require yaml
openroad> 
```

Also:

```
$ sta
OpenSTA 2.6.0 20925bb009 Copyright (c) 2024, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
sta [~/OpenROAD-flow-scripts/flow] package require yaml
error reading package index file /usr/lib/blt2.5/pkgIndex.tcl: Error: pkgIndex.tcl, 59 can't read "dir": no such variable
[deleted]
while evaluating package require yaml
sta [~/OpenROAD-flow-scripts/flow] 
```

### Expected Behavior

No error

### Environment

```shell
sta or OpenROAD
```

### To Reproduce

see above

donn, 2025-01-30:
This is the culprit:

https://github.com/The-OpenROAD-Project/OpenSTA/blob/ee700ba0f40e2a7101adc3ffd52953fc9f1f0245/sdc/Sdc.tcl#L109

OpenSTA hijacks `source` to add some conveniences when sourcing .sdc files. Unfortunately, that source function always runs every line at the top level.  dir, the variable in question, is not set at the top level, which causes this issue when attempting to load any Tcl package.

https://github.com/tcltk/tcl/blob/27f2debf9af32d287aea60d592107cea9f186718/library/package.tcl#L495

I'm not sure if or how you want to bring this up with OpenSTA upstream.

donn, 2025-01-30:
P.S. the fix is very simple, which is to just set the `uplevel` to 2 (to account for procs `source` and `source_`) instead of `#0`.

maliberty, 2025-01-30:
Please file against the upstream https://github.com/parallaxsw/OpenSTA

closed 2025-01-30 by maliberty

donn, 2025-01-31:
https://github.com/parallaxsw/OpenSTA/issues/202