The-OpenROAD-Project/OpenROAD #5877: Various errors when gain-based buffering used on sky130hd gcd or jpeg

widlarizer, 2024-10-04:
### Describe the bug

When building gcd, RCX at end of flow fails like so:

```
[INFO RCX-0436] RC segment generation gcd (max_merge_res 50.0) ...
[INFO RCX-0040] Final 1824 rc segments
[INFO RCX-0439] Coupling Cap extraction gcd ...
[INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
[INFO RCX-0442] 55% of 3751 wires extracted
[INFO RCX-0442] 100% of 3751 wires extracted
[INFO RCX-0045] Extract 646 nets, 2470 rsegs, 2470 caps, 3372 ccs
[INFO RCX-0443] 646 nets finished
[WARNING PSM-0038] Unconnected node on net VDD at location (64.857um, 37.775um), layer: met5.
[WARNING PSM-0039] Unconnected instance gain209/VPWR at location (64.857um, 37.775um).
[ERROR PSM-0069] Check connectivity failed on VDD.
Error: final_report.tcl, 61 PSM-0069
```

I then changed yosys from a dev version to the latest release. This changed the error to:

```
[INFO DRT-0195] Start 0th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:00:00, memory = 183.22 (MB).
    Completing 20% with 0 violations.
    elapsed time = 00:00:00, memory = 182.43 (MB).
    Completing 30% with 0 violations.
    elapsed time = 00:00:00, memory = 181.39 (MB).
    Completing 40% with 0 violations.
    elapsed time = 00:00:00, memory = 169.64 (MB).
    Completing 50% with 34 violations.
    elapsed time = 00:00:00, memory = 201.32 (MB).
Error: detail_route.tcl, 48 vector::_M_range_check: __n (which is 14) >= this->size() (which is 13)
```

This is an error I have seen in the same situation with jpeg.

### Expected Behavior

no errors

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.6.47
os: NixOS 24.11 (Vicuna)
cmake version 3.29.6
-- The CXX compiler identification is Clang 16.0.6
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /run/current-system/sw/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16235-gedf00dff9
-- System name: Linux
-- Compiler: Clang 16.0.6
-- Build type: RELEASE
-- Install prefix: /var/empty/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is Clang 16.0.6
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /run/current-system/sw/bin/cc - skipped
-- Detecting C compile features
[... 98 lines cut ...]
--- a/flow/scripts/floorplan.tcl
+++ b/flow/scripts/floorplan.tcl
@@ -2,6 +2,9 @@ utl::set_metrics_stage "floorplan__{}"
 source $::env(SCRIPTS_DIR)/load.tcl
 load_design 1_synth.v 1_synth.sdc

+remove_buffers
+repair_design -buffer_gain 2
+
```

and run

```
make REMOVE_ABC_BUFFERS=0 \
                                          DESIGN_CONFIG=designs/sky130hd/gcd/config.mk \
                                          EQUIVALENCE_CHECK=0 \
                                          finish
```

Personally I also set YOSYS_EXE and OPENROAD_EXE to have nothing in my path but you do you

### Additional Context

Once again this is with OpenROAD built with the flake, but that just uses standard tools

maliberty, 2024-10-07:
The drt issue comes from not using the DONT_USE_CELLS list and instantiating a probe cell (sky130_fd_sc_hd__probe_p_8).   The router should not error in this way but you should not use this cell.  Adding `set_dont_use $::env(DONT_USE_CELLS)` before you two new lines fixes it.

maliberty, 2024-10-07:
Please retest the RCX issue with whatever yosys version you are using.

maliberty, 2025-06-16:
Please reopen if still relevant