The-OpenROAD-Project/OpenROAD #5878: OpenROAD make command issue

Chudeerpalan, 2024-10-04:
### Describe the bug

[image]
no new designs added,default design nangate45/gcd/config.mk

### Expected Behavior

after entering make command ,the process should be started and floorplan will excuted 

### Environment

```shell
.
```

### To Reproduce

.

eder-matheus, 2024-10-04:
It seems that you don't have the binary for yosys. You can try running `./build_openroad.sh --local` in the root directory of the repository.

Chudeerpalan, 2024-10-05:
after doing ./build_openroad.sh inside openroad .. same issue

maliberty, 2024-10-05:
Have you run `. env.sh` to setup your path?

Chudeerpalan, 2024-10-05:
how to do

maliberty, 2024-10-05:
Have you read the sections on building in https://openroad-flow-scripts.readthedocs.io/en/latest/index2.html ?  I'm not sure if you are trying to build with docker or locally.

Chudeerpalan, 2024-10-07:
How to install open road in ubuntu 22.04 version locally. Can anyone share the top to bottom of installation guide.
I'm new to Linux system,so please share step by step procedure kindly.

eder-matheus, 2024-10-07:
> How to install open road in ubuntu 22.04 version locally. Can anyone share the top to bottom of installation guide. I'm new to Linux system,so please share step by step procedure kindly.

Here's the documentation about local installation available in OpenROAD-flow-scripts repository: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildLocally.md

Chudeerpalan, 2024-10-09:
Still I'm getting issue 
[makefile_error.pdf](https://github.com/user-attachments/files/17301700/makefile_error.pdf)
How to solve this problem.
I tired all the steps which is given in the documentation about local installation available in OpenROAD-flow-scripts repository

oharboe, 2024-10-09:
@Chudeerpalan Interesting error `Error: detail_route.tcl, 64 expected boolean value but got ""`, can you run `make detail_route_issue` and upload a reproducible case?

maliberty, 2024-10-09:
It seems you are past the yosys issue and now have a completely different problem.   When you built openroad did you use build.sh?  If so perhaps you can try with `--latest` to get the head of OpenROAD.

Chudeerpalan, 2024-10-09:
[image]

oharboe, 2024-10-09:
Great! Now upload that that .tar.gz file to github...

Chudeerpalan, 2024-10-09:
[makeIssue.sh.pdf](https://github.com/user-attachments/files/17304178/makeIssue.sh.pdf)

abdulwadood97, 2024-11-18:
I am also experiencing this issue. 
[WARNING GRT-0246] No diode with LEF class CORE ANTENNACELL found.
Error: detail_route.tcl, 61 expected boolean value but got ""
Command exited with non-zero status 1
Elapsed time: 0:10.76[h:]min:sec. CPU time: user 35.84 sys 0.65 (338%). Peak memory: 888760KB.
make[1]: *** [Makefile:769: do-5_2_route] Error 1
make: *** [Makefile:767: results/nangate45/gcd/base/5_2_route.odb] Error 2 

How to resolve? I ran make command in flow directory inside docker container.

closed 2025-06-16 by maliberty