The-OpenROAD-Project/OpenROAD #5882: Clicking on reg2reg bucket lists 0 paths in timing port

oharboe, 2024-10-05:
### Describe the bug

untar 
[reg2reg-bucket.tar.gz](https://github.com/user-attachments/files/17266840/reg2reg-bucket.tar.gz)

```
 ./run-me-ebtb_128x40-asap7-base.sh 
```

Navigate to reg2reg, there should be no paths in reg2reg, but there is a green bucket.

Clicking on the green bucket shows no timing paths(which I believe to be correct).

[image]

### Expected Behavior

If there are no paths, show no bucket, if there is a bucket and there are paths in it list only those paths

### Environment

```shell
OpenROAD v2.0-16257-g81b545063
```

### To Reproduce

See above

AcKoucher, 2024-10-10:
@maliberty  @oharboe 
If I run
```
report_checks -path_group reg2reg
```

I get
```
Startpoint: Memory[4][28]$_DFFE_PP_
            (rising edge-triggered flip-flop clocked by W0_clk)
Endpoint: Memory[4][28]$_DFFE_PP_
          (rising edge-triggered flip-flop clocked by W0_clk)
Path Group: reg2reg
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock W0_clk (rise edge)
  79.70   79.70   clock network delay (propagated)
   0.00   79.70 ^ Memory[4][28]$_DFFE_PP_/CLK (DFFHQNx2_ASAP7_75t_R)
  46.61  126.31 v Memory[4][28]$_DFFE_PP_/QN (DFFHQNx2_ASAP7_75t_R)
  16.94  143.24 ^ _4153_/Y (NOR2x1_ASAP7_75t_R)
  16.08  159.32 ^ _4154_/Y (AO21x1_ASAP7_75t_R)
   0.13  159.44 ^ Memory[4][28]$_DFFE_PP_/D (DFFHQNx2_ASAP7_75t_R)
         159.44   data arrival time

1500.00 1500.00   clock W0_clk (rise edge)
  73.14 1573.14   clock network delay (propagated)
   6.49 1579.62   clock reconvergence pessimism
        1579.62 ^ Memory[4][28]$_DFFE_PP_/CLK (DFFHQNx2_ASAP7_75t_R)
  -6.48 1573.14   library setup time
        1573.14   data required time
---------------------------------------------------------
        1573.14   data required time
        -159.44   data arrival time
---------------------------------------------------------
        1413.70   slack (MET)
```

This seems like a loop (?). 

I'm taking a look at why the report generated by clicking in the bucket shows nothing.

oharboe, 2024-10-10:
I dont think there is a loop in the Verilog, I dont understand this timing path....

maliberty, 2024-10-10:
Its a valid timing path.  Whether it should exist or not is outside the scope of this issue (would be a yosys thing most likely).

oharboe, 2024-10-10:
It would be good to understand, but mainly to satisfy curiosity....

PR #5920 (2024-10-13): gui: include pathgroup name information when searching for pathends
Resolve #5882

It seems like the manual check we were doing was wrong. There's an option to include the PathGroup(s) name(s) on the search so we can get only the paths that belong to that group.
closed 2024-10-13 by maliberty