The-OpenROAD-Project/OpenROAD #5885: add a "start_gui" .tcl command

oharboe, 2024-10-06:
### Description

The motivation is to use this to tweak gui.tcl to "warm up" the GUI to avoid long waiting times while the GUI is running.

Example use-case:

- a new variable, `export GUI_PRELOAD?=1` is added. When this is set, the GUI "warms up" before launching the GUI so that the GUI is more responsive once it opens up. This moves GUI waiting time into the "headless" phase, where running time is less performance critical(more about which further down).
- run `make gui_cts`
- runs gui.tcl and loads cts as usual
- executes `report_checks`, which will take a long time the first time.
- There are other things in the gui where the user has to wait until is calculated/loaded the first time. Heatmaps?

A further refinement is to use https://criu.org/ in combination with bazel and artifacts. criu can take a snapshot of the application state and restore it later. Add some ORFS tweaks to invoke criu.

With this in place, one can run:

- `make gui_cts GUI_CRCU_DUMP=somefolder` => saves a snapshot
- `make gui_cts GUI_CRCU_RESTORE=somefolder` => restores a snapshot

Add bazel support for the above with artifacts and one can launch a responsive GUI for a large design quickly.

I performed a smoke-test of criu and it works, though you'll want to avoid starting up the GUI before snapshotting openroad:

1. make open_cts, run "report_checks" => 5 minutes
2. sudo criu dump --no-file -t 89559 -D . --shell-job
(app is saved, then killed)
3. tar --use-compress-program=pigz -cvf ~/Desktop/xxx.tar.gz *.img => 2.5gBytes. odb file is 351mByte .tar.gz'ed.
4. sudo criu restore -D . --shell-job
5. seconds later, if I type "report_checks", it prints out the report

### Suggested Solution

implement start_gui tcl command

gadfort, 2024-10-06:
@oharboe all of this seems to be ORFS specific, did you mean to open this here?
From this description it's not clear what is needed in OpenROAD, as far as I can tell of these things can be handled in a tcl script in ORFS.

gadfort, 2024-10-06:
`gui::show`

oharboe, 2024-10-07:
Thanks! I checked that it works `make open_floorplan`, then type gui::show, opens the GUI.

closed 2024-10-07 by oharboe