The-OpenROAD-Project/OpenROAD #5893: code generator sections

andyfox-rushc, 2024-10-07:
### Describe the bug

The database code generator seems to have eccentric behaviour with "tag" names. eg meaningful names like "stmt1" seem to cause the code generator to have problems. For example if in odb/src/db/dbModBTerm.cpp at around line 187 the following User code block is specified:

  // User Code Begin stmt1

if (obj.getDatabase()->isSchema(db_schema_db_remove_hash)) {
    dbDatabase* db = (dbDatabase*) (obj.getDatabase());
    _dbBlock* block = (_dbBlock*) (db->getChip()->getBlock());
    _dbModule* module = block->_module_tbl->getPtr(obj._parent);
    if (obj._name) {
      module->_modbterm_hash[obj._name] = obj.getId();
    }
  }
  // User Code End stmt1

and error is issued. (Exception: User tag stmt1 not used in ../db/dbModBTerm.cpp)

Replace stmt1 with the symbol >> and things work fine.

Doubtless, pilot error, but it would be nice to be able to add meaningful tags to User Code blocks.

### Expected Behavior

generator should produce correct c++ code irrespective of name used for tag.

### Environment

```shell
./Env.sh
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.146.1-microsoft-standard-WSL2
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16290-gc4761c306
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
[... 131 lines cut ...]
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.oKujxCA5DL
```

### To Reproduce

For example if in odb/src/db/dbModBTerm.cpp at around line 187 the following User code block is specified:

  // User Code Begin stmt1

if (obj.getDatabase()->isSchema(db_schema_db_remove_hash)) {
    dbDatabase* db = (dbDatabase*) (obj.getDatabase());
    _dbBlock* block = (_dbBlock*) (db->getChip()->getBlock());
    _dbModule* module = block->_module_tbl->getPtr(obj._parent);
    if (obj._name) {
      module->_modbterm_hash[obj._name] = obj.getId();
    }
  }
  // User Code End stmt1

and error is issued. (Exception: User tag stmt1 not used in ../db/dbModBTerm.cpp)

Replace stmt1 with the symbol >> and things work fine.

maliberty, 2024-10-07:
There has to be some coordination between the template and the code.   The generator needs to know where to insert the user code.  If you look at https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/odb/src/codeGenerator/templates/serializer_in.cpp you'll see 
```
    //User Code Begin >>{{ comment_tag }}
    //User Code End >>{{ comment_tag }}
```
If we allowed arbitrary tag names then how would the generator know where to put the user code within the generated code?  In this case the tag name '>>' reflects that it is within `operator>>`.  Another name could be chosen but it needs to be selected at the template level not the user code and must be constant across all uses.

andyfox-rushc, 2024-10-07:
Please provide a simple example of what the tags can be.  The code generator simply needs to fetch the region marked by the tags and do nothing in those regions... Are you saying the tag names need to be something like //User Code Begin >> {{my_user_block1}} and the >> is some kind of streaming operator ??.  Right now the stuff following the Begin/End is unclear.

Presumably this is just a matter of finding the region between the User Code Begin and User Code End. But maybe I am missing something. One thing I am sure of, however, is that it is unclear what the tags should be and a few examples might help clarify.