The-OpenROAD-Project/OpenROAD #5905: pdngen get stuck

titan73, 2024-10-09:
### Describe the bug

pdngen get stuck after invocation.

```
read_lef test.lef

read_verilog test.v

link_design test

set db [ord::get_db_block]

set fp_area {0.000 0.000 2621.190 1028.475}

initialize_floorplan -die_area $fp_area -core_area $fp_area -site CORE

make_tracks

place_macro -macro_name instance -location [list 1823.130 105.280]

set master [[[$db findInst instance] getMaster] getName]

cut_rows -endcap_master $master -halo_width_x 9.400 -halo_width_y 9.400

add_global_connection -net VDD -pin_pattern {^VDD$} -power
add_global_connection -net GND -pin_pattern {^GND$} -ground

set_voltage_domain -power VDD -ground GND

define_pdn_grid -name "Core"

add_pdn_stripe -followpins -layer M1 -width 2.630

add_pdn_stripe -layer M3     -width 11.280 -pitch 225.600 -spacing 16.920 -offset 113.550
add_pdn_stripe -layer M4 -width 11.280 -pitch 526.400 -spacing 16.920 -offset 375.625

add_pdn_connect -layers {M3 M4}
add_pdn_connect -layers {M1 M3}

pdngen
```

### Expected Behavior

Should not get stuck

### Environment

```shell
OpenROAD v2.0-16249-gb095b8a8b
```

### To Reproduce

Run openroad test.tcl with test case provided.

[pdngen_freeze.zip](https://github.com/user-attachments/files/17311256/pdngen_freeze.zip)

maliberty, 2024-10-09:
When I try it I get a segv.  @gadfort do you have time to look at this?

gadfort, 2024-10-10:
I'll take a look this week. Do you have any details on the segv?

gadfort, 2024-10-10:
@titan73 I haven't worked out a fix yet, but the issue appears to be that M1 and M3 are supposed to connect, but they go in the same direction, which is causing some issues in the channel repair. If you change the connection from M1->M3 to M1->M4 it works.

Please leave this open as the segv @maliberty should be fixed.

titan73, 2024-10-10:
@gadfort Maybe the freeze is a different symptom of the same cause. The freeze disappear if I remove the macro by commenting these lines:

```
In test.v
    CELL instance ();

In test.tcl:
place_macro -macro_name instance -location [list 1823.130 105.280]
set master [[[$db findInst instance] getMaster] getName]
cut_rows -endcap_master $master -halo_width_x 9.400 -halo_width_y 9.400
```

In that case I get the error:

```
[WARNING PDN-0178] Remaining channel (0.0000, -1.3150) - (2621.0800, 1028.0650) on M1 for nets: VDD, GND
[ERROR PDN-0179] Unable to repair all channels.
```

Is this the error I'm supposed to that due to "add_pdn_connect -layers {M1 M3}" ?

titan73, 2024-10-10:
@gadford Btw, Why is a "channel" and what is "repairing"? I don't know what "Repairing channel" means.

titan73, 2024-10-10:
Tried with "add_pdn_connect -layers {M1 M4}" and I get (w/ & w/o macro):

[WARNING PDN-0178] Remaining channel (0.0000, 91.1850) - (2621.0800, 167.8150) on M1 for nets: VDD, GND
[WARNING PDN-0178] Remaining channel (0.0000, 313.1850) - (2621.0800, 389.8150) on M1 for nets: VDD, GND
[WARNING PDN-0178] Remaining channel (0.0000, 544.4350) - (2621.0800, 611.8150) on M1 for nets: VDD, GND
[WARNING PDN-0178] Remaining channel (0.0000, 766.4350) - (2621.0800, 843.0650) on M1 for nets: VDD, GND
[WARNING PDN-0178] Remaining channel (0.0000, 997.6850) - (2621.0800, 1028.0650) on M1 for nets: GND
[ERROR PDN-0179] Unable to repair all channels.
Error: test.tcl, 37 PDN-0179

gadfort, 2024-10-12:
A repair channel is a section of the power grid that could not be connected to with the configuration provided. It will attempt to adapt the configuration provided to repair these. However in your case they cannot be repaired since M1 and M3 are parallel layers.

PR #5927 (2024-10-12): pdn: add check for strap directions in repair channels, to avoid inf loop
Closes #5905 

The issue appears to be that parallel straps cannot be repaired, as the repair is expecting orthogonal straps to be able to find valid configurations. Without this, the code enters an infinite loop.
closed 2024-10-12 by maliberty