The-OpenROAD-Project/OpenROAD #5940: drt fails with `Error: checkConnectivity break, net sha512_inst.core.a_reg\[47\]`

gadfort, 2024-10-14:
### Describe the bug

Detail routing fails with
```
...
Error: checkConnectivity break, net sha512_inst.core.a_reg\[47\]
...
```

### Expected Behavior

connectivity check does not fail

### Environment

```shell
Commit: 7a67188f991813d986df6f7b180c1c15454292d6
```

### To Reproduce

[sc_issue_caliptra-sha512_job0_gf180_gf180mcu_fd_sc_mcu9t5v0_route0_20241014-013415.tar.gz](https://github.com/user-attachments/files/17364682/sc_issue_caliptra-sha512_job0_gf180_gf180mcu_fd_sc_mcu9t5v0_route0_20241014-013415.tar.gz)

```
tar xvf sc_issue_caliptra-sha512_job0_gf180_gf180mcu_fd_sc_mcu9t5v0_route0_20241014-013415.tar.gz
cd sc_issue_caliptra-sha512_job0_gf180_gf180mcu_fd_sc_mcu9t5v0_route0_20241014-013415
./run.sh
```

### Relevant log output

```shell
[INFO DRT-0195] Start 1st optimization iteration.
    Completing 10% with 78292 violations.
    elapsed time = 00:01:04, memory = 2512.68 (MB).
    Completing 20% with 78292 violations.
    elapsed time = 00:02:12, memory = 2512.68 (MB).
    Completing 30% with 61059 violations.
    elapsed time = 00:03:12, memory = 2512.68 (MB).
    Completing 40% with 61059 violations.
    elapsed time = 00:04:26, memory = 2514.68 (MB).
    Completing 50% with 61059 violations.
    elapsed time = 00:05:26, memory = 2524.73 (MB).
    Completing 60% with 43740 violations.
    elapsed time = 00:06:36, memory = 2530.04 (MB).
    Completing 70% with 43740 violations.
    elapsed time = 00:07:46, memory = 2540.09 (MB).
    Completing 80% with 25718 violations.
    elapsed time = 00:08:41, memory = 2550.54 (MB).
    Completing 90% with 25718 violations.
    elapsed time = 00:09:57, memory = 2553.02 (MB).
    Completing 100% with 7479 violations.
    elapsed time = 00:10:51, memory = 2553.02 (MB).
Error: sha512_inst.core.a_reg\[47\] 2 pin not visited #guides = 80
Error: checkConnectivity break, net sha512_inst.core.a_reg\[47\]
Objs not visited:
frPathSeg: begin (808640 2385600 ) end ( 820960 2385600 ) layerNum 2
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (805280 2385600 ) end ( 808640 2385600 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (805280 2385260 ) end ( 805280 2385600 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (810880 2587200 ) end ( 810880 2595040 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (810880 2587200 ) end ( 816480 2587200 ) layerNum 2
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (816480 2542400 ) end ( 820960 2542400 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (820960 2475200 ) end ( 820960 2542400 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (819840 2475200 ) end ( 820960 2475200 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (819840 2452800 ) end ( 819840 2475200 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (819840 2452800 ) end ( 820960 2452800 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (816480 2542400 ) end ( 816480 2587200 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (820960 2385600 ) end ( 820960 2452800 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frVia: at ( 820960 2385600 )
VIA DEF:
VIA Via1_HV DEFAULT
  RECT -380 -260 380 260
  RECT -260 -260 260 260
  RECT -280 -380 280 380
frVia: at ( 808640 2385600 )
VIA DEF:
VIA Via1_HV DEFAULT
  RECT -380 -260 380 260
  RECT -260 -260 260 260
  RECT -280 -380 280 380
frVia: at ( 808640 2385600 )
VIA DEF:
VIA Via2_VH DEFAULT
  RECT -280 -380 280 380
  RECT -260 -260 260 260
  RECT -380 -280 380 280
frVia: at ( 805280 2385600 )
VIA DEF:
VIA Via2_VH DEFAULT
  RECT -280 -380 280 380
  RECT -260 -260 260 260
  RECT -380 -280 380 280
frVia: at ( 805280 2385260 )
VIA DEF:
VIA Via1_HV DEFAULT
  RECT -380 -260 380 260
  RECT -260 -260 260 260
  RECT -280 -380 280 380
frVia: at ( 810880 2595040 )
VIA DEF:
VIA Via1_HV DEFAULT
  RECT -380 -260 380 260
  RECT -260 -260 260 260
  RECT -280 -380 280 380
frVia: at ( 810880 2587200 )
VIA DEF:
VIA Via1_HV DEFAULT
  RECT -380 -260 380 260
  RECT -260 -260 260 260
  RECT -280 -380 280 380
frVia: at ( 816480 2587200 )
VIA DEF:
VIA Via1_HV DEFAULT
  RECT -380 -260 380 260
  RECT -260 -260 260 260
  RECT -280 -380 280 380
INSTTERM: (INST/CELL/TERM/NET) _26624_ gf180mcu_fd_sc_mcu9t5v0__xnor3_2 A1 sha512_inst.core.a_reg\[47\]
```

osamahammad21, 2024-10-14:
routes cleanly with TR-guides-on-aps. will close when it's merged.

gadfort, 2024-10-14:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/5933 this one?

osamahammad21, 2024-10-14:
yup

osamahammad21, 2024-10-15:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/5933 is merged

closed 2024-10-15 by osamahammad21