The-OpenROAD-Project/OpenROAD #5952: pdn: Failure due to missing follow pins vias

titan73, 2024-10-15:
### Describe the bug

pdngen fails with the errors in log below.

The pdn script part is:
```
add_global_connection -net VDD -pin_pattern {^VDD$} -power
add_global_connection -net GND -pin_pattern {^GND$} -ground

set_voltage_domain -power VDD -ground GND

define_pdn_grid -name "Core"

add_pdn_stripe -followpins -layer M1 -width 2.630

add_pdn_stripe -layer M3     -width 11.280 -pitch 924.960 -spacing 16.920 -offset 37.000
add_pdn_stripe -layer M4 -width 11.280 -pitch 723.800   -spacing 16.920 -offset 107.910

add_pdn_connect -layers {M3 M4}
add_pdn_connect -layers {M1 M4}

pdn::debug_renderer 1
pdn::debug_renderer_update

pdngen
```

With the debug feature, we can see the missing vias:

[image]

[image]

### Expected Behavior

No missing vias and pdn created succesfully.

### Environment

```shell
Latest OR
```

### To Reproduce

openroad test.tcl -gui with following test case:

[missing_fp_vias.zip](https://github.com/user-attachments/files/17381260/missing_fp_vias.zip)

### Relevant log output

```shell
[INFO PDN-0001] Inserting grid: Core
[WARNING PDN-0178] Remaining channel (0.0000, 17.1850) - (2621.0800, 84.5650) on M1 for nets: VDD, GND
[WARNING PDN-0178] Remaining channel (0.0000, 942.1850) - (1812.5800, 1009.5650) on M1 for nets: VDD, GND
[ERROR PDN-0179] Unable to repair all channels.
[ERROR GUI-0070] Error: test.tcl, 40 PDN-0179
```

gadfort, 2024-10-15:
@titan73 It's not clear to me what the M3 strap is doing for you, but it's the cause of this issue.
```
#add_pdn_stripe -layer M3     -width 11.280 -pitch 924.960 -spacing 16.920 -offset 37.000
add_pdn_stripe -layer M4 -width 11.280 -pitch 723.800   -spacing 16.920 -offset 107.910

#add_pdn_connect -layers {M3 M4}
add_pdn_connect -layers {M1 M4}
```

Without the M3 strap it compiles without issue. If you are worried about IR drop you can always reduce the pitch on M4.

titan73, 2024-10-15:
@gadfort The pictures show only the lower left part of the floorplan to show the details. There are several M4 strap P/G pairs and only 2 pins for power & ground (not in the test case) so these M4 straps need to be connected together with M3 ones.

gadfort, 2024-10-15:
Can you provide a complete testcase then? With what is provided, removing the M3 strap is sufficient.

titan73, 2024-10-15:
I can't provide the full test case for legal reasons and we don't have support contract with PII yet. The request has been done to my hierarchy but it will be difficult since it's evaluation work. That's why I made a simplified one.
That said removing M3 straps is a workaround. Having a power grid using 2 metals is not something rare. The problem comes from the fact that M3 has the same direction as M1 so M1 needs to connect to M4 and pdn does not handle the M3 blocking the way.

gadfort, 2024-10-16:
If you must have the M3 straps, then you can add back:
```
add_pdn_connect -layers {M1 M3}
```

That will complete the power grid, but because the M3 and M1 overlap completely you will have one giant via array and a massive routing blockage on M2.
To limit the size of M2 you could use of a routing blockage, however the via code will reject the via due to the obstruction. This could be fixed however, but will require some time.

titan73, 2024-10-16:
Ok. With this command, I have a power grid closer to the expected one. Thanks.
That could be good to have a fix. No hurry though.

titan73, 2024-10-16:
I didn't pay attention but with the 2 lines:
```
add_pdn_connect -layers {M1 M3}
add_pdn_connect -layers {M1 M4}
```

This create a barrier that prevent routing vertically on M2 on top & bottom edges (M4 is too thick and not very used for rooting):

[image]

I'll comment out pdngen for now then.