The-OpenROAD-Project/OpenROAD #5953: `repair_clock_nets` does not honor max_fanout

gadfort, 2024-10-15:
### Describe the bug

After running `repair_clock_nets` there are new max_fanout violations.

### Expected Behavior

No max fanout violations

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.0-45-generic
os: Ubuntu 24.04.1 LTS (Noble Numbat)
cmake version 3.24.2
-- The CXX compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16487-g0e060db59
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
[... 111 lines cut ...]
-- Configuring done
-- Generating done
```

### To Reproduce

[sc_issue_tiny_rocket_job0_ihp130_sg13g2_stdcell_cts0_20241015-121405.tar.gz](https://github.com/user-attachments/files/17381935/sc_issue_tiny_rocket_job0_ihp130_sg13g2_stdcell_cts0_20241015-121405.tar.gz)

```
ta xvf sc_issue_tiny_rocket_job0_ihp130_sg13g2_stdcell_cts0_20241015-121405.tar.gz
cd sc_issue_tiny_rocket_job0_ihp130_sg13g2_stdcell_cts0_20241015-121405
./run.sh
```

### Relevant log output

```shell
max fanout

Pin                                   Limit Fanout  Slack
---------------------------------------------------------
clkbuf_leaf_411_clock/X                   8      9        (VIOLATED)
clkbuf_leaf_418_clock/X                   8      9        (VIOLATED)
clkbuf_leaf_428_clock/X                   8      9        (VIOLATED)
```

maliberty, 2024-10-15:
I believe @arthurjolo is looking at this already - please update.

maliberty, 2024-10-15:
I believe the goal is for CTS to not leave such violations to repair_clock_nets

gadfort, 2024-10-15:
@arthurjolo yeah, checking before the call to repair_clock_nets there is not an issue, so it's something that is added by `repair_clock_nets`

arthurjolo, 2024-10-15:
@gadfort I looked at this case and in fact CTS was the one that was causing the violations. They where happening due to the dummy load insertion, that did not considered the fanout limit and was inserting buffers to nets that the driver already was on the fanout limit, I just opened a PR to fix this.

closed 2024-10-16 by maliberty