The-OpenROAD-Project/OpenROAD #5962: Add a `macro_place -legalize` option for use with approximate manual placement

oharboe, 2024-10-16:
### Description

Approximate manual placement is easy, but the rules with pins on multiple metal layers for manual placement are tricky.

### Suggested Solution

Add a `macro_place -legalize` option for use with approximate manual placement

oharboe, 2024-10-16:
> @oharboe You can use the `place_macro` command to place individual macros. It should automatically legalize the macro with regards to the pins.

Understood, but what I'm after is to legalize some approximate positions.

oharboe, 2024-10-16:
Perhaps by giving in an approximate position to place_macro?

oharboe, 2024-10-16:
@jeffng-or has a usecase where he wants to move one macro a bit to the right to see if it works better in global route:

1. write_macro_placement
2. add 100um to one x position
3. add a -legalize arg to the macro_place that was adjusted

maliberty, 2024-10-16:
Did you meant place_macro in the title?

oharboe, 2024-10-16:
> Did you meant place_macro in the title?

I did not consider that there were two commands...

As a user, I have no strong opinion about syntax.

maliberty, 2024-10-16:
macro_place is the old macro placer and not for user given placement.  I think that you mean place_macro -legalize.

eder-matheus, 2024-10-16:
When using the write_macro_placement command, it will generate a file with multple calls to the place_macro command. If you change the  x/y position in that file and execute it, it will already place the macro in a valid position, as far as I know. This is very similar to the write_pin_placement command in ppl.

maliberty, 2024-10-16:
It will place the macro where you say.  If you pick a location where the pins are off the routing tracks that's undesirable and painful to calculate manually.  I think -legalize would just call the snapper.

AcKoucher, 2024-10-16:
> If you pick a location where the pins are off the routing tracks that's undesirable and painful to calculate manually. I think -legalize would just call the snapper.

Unless the user is using a right angle rotation orientation, the snapper will be called in `place_macro`.

maliberty, 2024-10-16:
@oharboe perhaps it already does what you need ?

AcKoucher, 2024-10-16:
Close?

oharboe, 2024-10-16:
Will open test case if it does not work as advertised here
👍

closed 2024-10-16 by oharboe