The-OpenROAD-Project/OpenROAD #5971: Placement results in path that appears to lap macro

jeffng-or, 2024-10-16:
### Describe the bug

In megaboom v6, the resulting floorplan/macro/std cell placement results in paths that appear to "lap" one of the macros multiple times, which intuitively seems wrong:

[image]

### Expected Behavior

I'm not sure where the exact issue is, so hopefully the development team can provide insight into it.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.0-1025-gcp
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16316-gf9cfd9383
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
[... 101 lines cut ...]
-- FounNumber of processor cores: 32
d Boost: /home/jeffng/dev/main/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /home/jeffng/dev/main/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /home/jeffng/dev/main/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /home/jeffng/dev/main/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- Found Eigen3: /home/jeffng/dev/main/OpenROAD-flow-scripts/dependencies/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.Bt0W42tFUE
```

### To Reproduce

1. unpack the tarball https://drive.google.com/file/d/1eVCheqXsRstEgBQLqIi-H5i4qwO9Z6Dh/view?usp=sharing
2. cd cts_BoomTile_asap7_base_2024-10-16_10-40
3. ./run-me-BoomTile-asap7-base.sh gui <- I added the second argument to just pull up the GUI instead of running CTS
4. type gui::show if the GUI doesn't come up (might be fixed)
5. click Update on the Timing Report
6. Pick the first path (although any of the ones with -6k slack would suffice

maliberty, 2024-10-16:
At what point in the flow is your image from?  Trying to narrow down where in the flow the problem starts is helpful.

stefanottili, 2024-10-16:
Give the “visualize the whole cone” feature a try. I found it very useful to explain why placers put cells in unintuitive locations. There are usually more things pulling at cells than you see by just looking at the critical path.

jeffng-or, 2024-10-16:
> At what point in the flow is your image from? Trying to narrow down where in the flow the problem starts is helpful.

This is after the CTS step

maliberty, 2024-10-16:
Are you using global placement with -timing_driven?  If not you should turn it on.

jeffng-or, 2024-10-17:
> Are you using global placement with -timing_driven? If not you should turn it on.

No, the bazel var GPL_TIMING_DRIVEN is set to 0. I can run it with the option to see what changes.

jeffng-or, 2024-10-17:
With GPL_TIMING_DRIVEN enabled, the previous WC path improved -1673 slack vs -6393. Here's what the path looks like now:

[image]

So, it's still looking like it's taking a lap around the macro.

The fanout cone visualization looks like:

[image]

FWIW, the new WC path is -1930 slack.

stefanottili, 2024-10-17:
The fan out cone needs some more refinement to be really useful:
1) draw the flight lines bottom up, from least to most critical
2) get an adjustable filter to get rid of the not so critical ones ..
But it’s the only way that I know to visualize both connectivity and timing criticality.

maliberty, 2024-10-17:
The data is organized by level rather than by criticality currently.   We use the top 1000 paths to the pin of interest.  Do you have a suggestion for a good default filter until it can be made user selectable?

stefanottili, 2024-10-17:
Anything that reduces the amount of data to a point that it's "fast to draw" and "useful" would help, so "user selectable level" would be nice, but criticality is obviously better.

Even when there was a tight collaboration between RTL writer and P&R tool jockey, I was always surprised by the lack of knowledge about the "size of the cones", both # and location of starting/ending ff's of a cone and the routing that goes with it. There are only few RTL guy's (architects) that take the time to "to have a look" at how their netlist is placed.

So anything that gives insight into why things end up where they are and "what else is critical" in the vicinity is helpful.

No P&R tool that I'm aware of has "a nice way" to visualize "input/output cones color coded by criticality", so OR's cone viewing is a great first step, it just needs widespread usage and user feedback to get the proper refinement.
Useful visualization is more than "just" showing the one critical path and the clk skew at its input and output ff.

jeffng-or, 2024-10-29:
@precisionmoon's input:

All the cells in report_checks are std cells.   There are buffers added by repair_design but their placements are already impacted by pre-placed non-buffer cells.
Timing-driven option seems to be the way to go.  -timing_driven_net_reweight_overflow can be changed to decrease runtime.

oharboe, 2024-11-19:
More examples https://github.com/The-OpenROAD-Project/megaboom/pull/208

[image]