The-OpenROAD-Project/OpenROAD #5973: grt::have_routes Signal 11 received

oharboe, 2024-10-17:
### Describe the bug

untar [detail_route_gcd_nangate45_base_2024-10-17_07-06.tar.gz](https://github.com/user-attachments/files/17406325/detail_route_gcd_nangate45_base_2024-10-17_07-06.tar.gz)

```
$ ./run-me-gcd-nangate45-base.sh 
OpenROAD v2.0-16537-gb71907cde 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Signal 11 received
Stack trace:
 0# 0x00005E924D1A8CE3 in openroad
 1# 0x000076EBA9245320 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbChip::getBlock() in openroad
 3# grt::GlobalRouter::initGridAndNets() in openroad
 4# 0x00005E924F49980C in openroad
 5# grt::GlobalRouter::haveRoutes() in openroad
 6# 0x00005E924D9214FA in openroad
 7# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# 0x000076EBAE2B96BC in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
11# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
12# 0x00005E924D1A85B6 in openroad
13# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# main in openroad
15# 0x000076EBA922A1CA in /lib/x86_64-linux-gnu/libc.so.6
16# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
17# _start in openroad
./run-me-gcd-nangate45-base.sh: line 7: 786576 Segmentation fault      (core dumped) openroad -no_init ${SCRIPTS_DIR}/detail_route.tcl
```

### Expected Behavior

no crash

### Environment

```shell
OpenROAD v2.0-16537-gb71907cde
```

### To Reproduce

See above

eder-matheus, 2024-10-17:
@oharboe In your script you're calling grt::have_routes before loading the db file. I created a PR that will throw an error message in that case, avoiding the crash. But just moving the load of the db file should fix the issue in your end.

oharboe, 2024-10-17:
> @oharboe In your script you're calling grt::have_routes before loading the db file. I created a PR that will throw an error message in that case, avoiding the crash. But just moving the load of the db file should fix the issue in your end.

Yes: fixed in my script. Thanks!

closed 2024-10-20 by maliberty