The-OpenROAD-Project/OpenROAD #5975: drt; no access point issue

titan73, 2024-10-17:
### Describe the bug

No access point and routing fails.

### Expected Behavior

Find access points for 2 cell pins and route.

### Environment

```shell
v2.0-16472-gc7283c089
```

### To Reproduce

Test case with 1 buffer and 2 ports:
[no_access_point.zip](https://github.com/user-attachments/files/17412706/no_access_point.zip)

### Relevant log output

```shell
[INFO DRT-0167] List of default vias:
  Layer V1
    default via: M2_M1_VH_via
[INFO DRT-0162] Library cell analysis.
[INFO DRT-0163] Instance analysis.
[INFO DRT-0164] Number of unique instances = 1.
[INFO DRT-0168] Init region query.
[INFO DRT-0024]   Complete FR_MASTERSLICE.
[INFO DRT-0024]   Complete Fr_VIA.
[INFO DRT-0024]   Complete M1.
[INFO DRT-0024]   Complete V1.
[INFO DRT-0024]   Complete M2.
[INFO DRT-0033] FR_MASTERSLICE shape region query size = 0.
[INFO DRT-0033] FR_VIA shape region query size = 0.
[INFO DRT-0033] M1 shape region query size = 15.
[INFO DRT-0033] V1 shape region query size = 0.
[INFO DRT-0033] M2 shape region query size = 2.
[INFO DRT-0165] Start pin access.
[ERROR DRT-0073] No access point for instance/A.
Error: test.tcl, 21 DRT-0073
```

maliberty, 2024-10-17:
@bnmfw you might start with the debug graphics to see what is being tried: Add `detailed_route_debug -pa -pa_markers -pin instance:A` and run with the GUI.

titan73, 2024-12-11:
@bnmfw Any news on this issue?

jeffng-or, 2024-12-11:
FYI. Also seeing this with MegaBoom when M3/M4 routing layer adjustment is set to 0.3.
```
[INFO DRT-0165] Start pin access.
[INFO DRT-0076]   Complete 1000 pins.
[INFO DRT-0076]   Complete 2000 pins.
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/btb/btb_0_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/btb/btb_1_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/tage/t_2/lo_us_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/tage/t_2/hi_us_ext/R0_addr[0].
[ERROR DRT-0073] No access point for dcache/mshrs/lb_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/btb/meta_1_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/btb/meta_0_ext/R0_addr[0].
[ERROR DRT-0073] No access point for dcache/mshrs/sdq_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/tage/t_1/table_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_1/tage/t_2/table_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/tage/t_4/table_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/tage/t/table_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/bpd/banked_predictors_0/tage/t_2/table_ext/R0_addr[0].
[ERROR DRT-0073] No access point for frontend/icache/tag_array_ext/RW0_addr[0].
```

poudelbidhan, 2025-02-18:
@titan73 any new update on the issue ? were you able to figure out what was causing the issue or any workaround to solve it ?

maliberty, 2025-02-19:
The test case is slightly off ask make_tracks needs before place_pins.   The real issue is that the pin is 0.33 which triggers 
```
  MINIMUMCUT 2 WIDTH 0.310 FROMABOVE ;
```
OR doesn't support using double cut vias today.  You could mark this cell don't use but I don't know how common this is in your library.

maliberty, 2025-02-19:
[image]

maliberty, 2025-02-19:
I don't even see a double check via in your LEF to use.

titan73, 2025-02-20:
Thanks for the analysis.

> The test case is slightly off ask make_tracks needs before place_pins.

Yeah this is required now.

> The real issue is that the pin is 0.33 which triggers
> 
> ```
>   MINIMUMCUT 2 WIDTH 0.310 FROMABOVE ;
> ```
> 

You mean MINIMUMCUT 2 WIDTH 0.250 FROMABOVE ; ?

> OR doesn't support using double cut vias today. You could mark this cell don't use but I don't know how common this is in your library.

All cells are this way.
According to the lef, double cut vias are always needed for M1:
```
LAYER M1
  ...
  WIDTH 0.330 ;
  ...
  MINIMUMCUT 2 WIDTH 0.250 FROMABOVE ;
  ...
END M1
```

So double cut support is needed for this techno.

titan73, 2025-02-20:
> I don't even see a double check via in your LEF to use.

Yeah indeed, my test case building script mistakenly strip them out.

maliberty, 2025-02-20:
> So double cut support is needed for this techno.

Sorry but the router will not support this technology currently.

titan73, 2025-04-25:
It appears that the rule:
`MINIMUMCUT 2 WIDTH 0.310 FROMABOVE ;`
is invalid because the specified width is lower than the minimum width or default width:
`  WIDTH 0.315 ;`
Innovus ignore them during routing (cts, route and postroute):
#WARNING (xxx) Ignoring layer metal1 MINIMUMCUT rule with WIDTH (0.310000) <= the layer's MINWIDTH (0.315000).

@VoltXd, our student, has written a scrip code to preprocessed techno lef to strip out these invalid rules.

That would be good to ignored these invalid rules in OR instead of preprocessing techno lef? What could be the best location for doing this? LEF parser appears to be the best place for that. What do you think?