The-OpenROAD-Project/OpenROAD #5980: grt-226 error during antenna repair

gadfort, 2024-10-18:
### Describe the bug

When performing antenna repair I'm getting:
```
[ERROR GRT-0226] Type2 ripup not type L.
```

I was not getting this error when running global routing and antenna repair as part of the same session. The only thing that changes was saving the database to disk and loading it for antenna repair.

### Expected Behavior

No failure

### Environment

```shell
OpenROAD v2.0-16595-g2f10d9354 
Features included (+) or not (-): +Charts +GPU +GUI +Python
```

### To Reproduce

[sc_issue_picorv32_job0_route.antenna_repair0_20241017-200804.tar.gz](https://github.com/user-attachments/files/17425202/sc_issue_picorv32_job0_route.antenna_repair0_20241017-200804.tar.gz)

```
tar xvf sc_issue_picorv32_job0_route.antenna_repair0_20241017-200804.tar.gz
cd sc_issue_picorv32_job0_route.antenna_repair0_20241017-200804
./run.sh
```

### Relevant log output

```shell
[INFO ANT-0002] Found 11 net violations.
[INFO ANT-0001] Found 13 pin violations.
[INFO GRT-0012] Found 11 antenna violations.
[INFO GRT-0015] Inserted 13 diodes.
[ERROR GRT-0226] Type2 ripup not type L.
```

gadfort, 2024-10-18:
@eder-matheus thanks for clarifying it would be nice to be able to support it if possible

gadfort, 2024-10-21:
@eder-matheus does this also mean that incremental grt is also not supported or post drt antenna repair?

eder-matheus, 2024-10-21:
> @eder-matheus does this also mean that incremental grt is also not supported or post drt antenna repair?

Post-GRT timing repair should work when reading the an odb file. Post-DRT should also work since it uses the odb wires to detect violations and to calculate the available resources in the design.

Only repair antennas from guides in odb that is not supported yet.

closed 2024-10-22 by maliberty