The-OpenROAD-Project/OpenROAD #5981: Remove Static Variables

LucasYuki, 2024-10-18:
### Description

OpenROAD shouldn't use any static variables under the hood that aren’t somehow guarded by a lock.

Static variables per module:
- [ ] cts - 1 static variable
- [ ] drt - 17 static variables
- [ ] gui - 12 static variables
- [ ] odb - 108 static variables
- [ ] rcx - 8 static variables
- [ ] rmp - 1 static variable
- [ ] sta - 166 static variables
- [ ] stt - 7 static variables

maliberty, 2024-10-18:
sta is special so let's address it last and discuss first.

maliberty, 2024-11-18:
```
odb::db_tbl
odb::db_unique_id
```
fixed in #6192

maliberty, 2024-11-18:
`odb::conn` fixed in #6190

maliberty, 2024-11-18:
`odb::ttttsv` fixed in #6191

maliberty, 2024-11-18:
```
log_filename
metrics_filename
```
fixed in #6194

maliberty, 2024-11-18:
`cts::HTreeBuilder::plotSolution()::cnt` fixed in #6195

maliberty, 2024-11-18:
@LucasYuki no need to do gui (excluded from the current plan).

maliberty, 2024-11-18:
@QuantamHD agreed to handle the one in rmp (discussed in #5989)

maliberty, 2025-02-21:
sta to be managed separately

closed 2025-02-21 by maliberty