The-OpenROAD-Project/OpenROAD #5983: Lots of macro positions in macro place in GUI and in scripts - none of them match up

oharboe, 2024-10-18:
### Describe the bug

untar [macro-place.tar.gz](https://github.com/user-attachments/files/17435301/macro-place.tar.gz)

```
./run-me-mock-array-asap7-base.sh
```

In GUI, run:

```
source macros.tcl
```

macros.tcl was written out by write_macro_placement for `make DESIGN_CONFIG=designs/asap7/mock-array/config.mk gui_2_4_macro_place`

The last line of macros.tcl is:

```
place_macro -macro_name ces_7_7 -location {309.024 339.264} -orientation R0
```

All of the macro positions for ces_7_7 that I see are different, I have no idea why:

macros.tcl 309.024 339.264

[image]

### Expected Behavior

I have no idea what these macro positions are and what they should be, so I don't know what to make of it or what to expect.

### Environment

```shell
v2.0-16600-gb972b3473
```

### To Reproduce

See above

maliberty, 2024-10-19:
See https://github.com/The-OpenROAD-Project/OpenROAD/issues/5962

oharboe, 2024-10-19:
So pin positions are rounded, but why isn't write_macro_placement writing out the snapped(adjusted) positions then?

PR #5990 (2024-10-21): mpl2: don't move macro if all layers are already snapped
Fix #5983 

Also: use new block methods when writing mpl string.