The-OpenROAD-Project/OpenROAD #5985: write_abstract_lef uses Block bounding box instad of Die Area of `SIZE`

smunaut, 2024-10-19:
### Describe the bug

When writing out an abstract LEF, the code in `src/odb/src/lefout/lefout.cpp` / `lefout::writeBlock` used the `db_block->getBBox()` to select what `SIZE` to write in the LEF.

The bounding box will be dependent on the content of the macro and the defined pins and such which can change if some things are changed when re-hardening the macro.

This is really annoying because when instanciating the macro using a different orientation than `N`, because the position specified is the one of the lower left corner after orientation is applied, rather than the position of the origin of the macro. This means when the LEF SIZE changes, the position of the pins changes for whoever instanciate the macro ..

### Expected Behavior

What I'd expect is for the LEF SIZE to be the Die Size which is user selectable during floorplan and is stable.

```diff
diff --git a/src/odb/src/lefout/lefout.cpp b/src/odb/src/lefout/lefout.cpp
index 2b3c18d9c..c3dfd7d51 100644
--- a/src/odb/src/lefout/lefout.cpp
+++ b/src/odb/src/lefout/lefout.cpp
@@ -497,9 +497,9 @@ void lefout::writeBlockVia(dbBlock* db_block, dbVia* via)
 
 void lefout::writeBlock(dbBlock* db_block)
 {
-  dbBox* bounding_box = db_block->getBBox();
-  double size_x = lefdist(bounding_box->xMax());
-  double size_y = lefdist(bounding_box->yMax());
+  Rect die_area = db_block->getDieArea();
+  double size_x = lefdist(die_area.dx());
+  double size_y = lefdist(die_area.dy());
 
   for (auto via : db_block->getVias()) {
     writeBlockVia(db_block, via);
```

### Environment

```shell
OpenROAD v2.0-15789-gc38fd8ad1
```

### To Reproduce

```
read_db any_random_project.odb`
write_abstract_lef /tmp/test.lef
write_def /tmp/test.def
```

Observe that the size in the LEF is not the DIE AREA of the DEF.

maliberty, 2024-10-19:
I suppose this only comes up if you have no pins on the right / top edge otherwise they would force the bbox==die area.  Can you make a PR?

smunaut, 2024-10-19:
Yes, indeed it happened on our tiny tapout tiles that have pins only on the top.

I'll open a PR, I'm just double checking the LEF specs to see what to do if die area lower left is not 0.0 / 0.0.

maliberty, 2024-10-19:
I think you should assume it is 0,0 as OR doesn't support other non-standard choices.

closed 2024-10-19 by maliberty