The-OpenROAD-Project/OpenROAD #5988: [WARNING RSZ-0075] makeBufferedNet failed for driver core/rob/core/rob/_99958_/CON

oharboe, 2024-10-20:
### Describe the bug

untar https://drive.google.com/file/d/1Yhg6RdbrmuNTAFz9-th3IIC2ZCWkrhNi/view?usp=sharing

```
./run-me-BoomTile-asap7-2.sh
```

```
OpenROAD v2.0-16664-g9ba039016
[deleted]
Running floorplan.tcl, stage 2_1_floorplan
[INFO ODB-0227] LEF file: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7_tech_1x_201209.lef, created 24 layers, 9 vias
[INFO ODB-0227] LEF file: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef, created 212 library cells
[INFO ODB-0227] LEF file: results/asap7/array_256x128/base/array_256x128.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/btb_128x56/base/btb_128x56.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/dataArrayB_256x64/base/dataArrayB_256x64.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/data_2048x8/base/data_2048x8.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/ebtb_128x40/base/ebtb_128x40.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/lb_32x128/base/lb_32x128.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/mem_256x4/base/mem_256x4.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/meta_128x120/base/meta_128x120.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/meta_40x240/base/meta_40x240.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/sdq_17x64/base/sdq_17x64.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/table_128x44/base/table_128x44.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/table_128x52/base/table_128x52.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/table_256x48/base/table_256x48.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/tag_array_64x168/base/tag_array_64x168.lef, created 5 vias, 1 library cells
[INFO ODB-0227] LEF file: results/asap7/tag_array_64x184/base/tag_array_64x184.lef, created 5 vias, 1 library cells

==========================================================================
Floorplan check_setup
--------------------------------------------------------------------------
Warning: There are 148 input ports missing set_input_delay.
Warning: There are 238 output ports missing set_output_delay.
Warning: There are 684 unconstrained endpoints.
number instances in verilog is 2043016
[WARNING IFP-0028] Core area lower left (2.000, 2.000) snapped to (2.052, 2.160).
[INFO IFP-0001] Added 5540 rows of 27702 site asap7sc7p5t.
repair_timing -verbose -repair_tns 0 -skip_last_gasp
[WARNING RSZ-0021] no estimated parasitics. Using wire load models.
[INFO RSZ-0094] Found 165887 endpoints with setup violations.
[INFO RSZ-0099] Repairing 1 out of 165887 (0.00%) violating endpoints...
   Iter   | Removed | Resized | Inserted | Cloned |  Pin  |    WNS   |   TNS      |  Viol  | Worst
          | Buffers |  Gates  | Buffers  |  Gates | Swaps |          |            | Endpts | Endpt
---------------------------------------------------------------------------------------------------
        0 |       0 |       0 |        0 |      0 |     0 | -2195.436 | -114376600.0 | 165887 | core/int_issue_unit/core/int_issue_unit/io_dis_uops_0_ready_REG$_DFF_P_/D
       10 |       0 |       7 |        0 |      1 |     1 | -1346.351 | -80619768.0 | 165887 | core/int_issue_unit/core/int_issue_unit/io_dis_uops_0_ready_REG$_DFF_P_/D
[WARNING RSZ-0075] makeBufferedNet failed for driver core/rob/core/rob/_99958_/CON
[WARNING RSZ-0075] makeBufferedNet failed for driver core/rob/core/rob/_99958_/CON
[WARNING RSZ-0075] makeBufferedNet failed for driver core/rob/core/rob/_99958_/CON
[deleted]
```

### Expected Behavior

No warning

### Environment

```shell
OpenROAD v2.0-16664-g9ba039016
```

### To Reproduce

See above

oharboe, 2024-10-23:
@maliberty Is this just a warning that should be fixed or does it have any detrimental effect relevant to megaboom?

maliberty, 2024-10-23:
Mostly a warning.  In short it is saying it can't insert buffers into nets pre-placement.  Eventually that will be good to fix but for now we will just disable the attempt and therefor the warning.