The-OpenROAD-Project/OpenROAD #5999: inspector back arrow doesn't clear highlighting of modules

maliberty, 2024-10-22:
### Describe the bug

I'm looking at ng45/ariane136.  If I select an instance I see:
[image]

I then click the Module  'i_frontend':
[image]

Then clicking the back arrow:
[image]

The inspected object is correct but the highlighting lingers incorrectly.

### Expected Behavior

Highlighting in the last image should match the first.

### Environment

```shell
Head OR.
```

### To Reproduce

Should happen on any design with verilog hierarchy.

maliberty, 2024-10-22:
@gadfort would you have time to look at this?  Not urgent.

gadfort, 2024-10-22:
@maliberty sure, I can give it a stab

PR #6023 (2024-10-25): gui: fix selection in inspector when using navigation
Fixes #5999

Changes:
- make qDebug messages go to `debugPrint` to allow for easier QT debugging.
closed 2024-10-25 by maliberty