The-OpenROAD-Project/OpenROAD #6030: Speed up report_metrics in CTS

oharboe, 2024-10-27:
### Description

Example:

untar https://drive.google.com/file/d/1kSGyQaVFeZ94_s95Ce2tG80UHuYcR6fi/view?usp=sharing

./run-me-BoomTile-asap7-3.sh

setting SKIP_REPORT_METRICS=0 takes down running time from approximately 3000 to 600 seconds

### Suggested Solution

Make it faster

maliberty, 2024-10-27:
Would you break down the runtime difference by step?

oharboe, 2024-10-27:
> Would you break down the runtime difference by step?

It is all in cts, in the included example, which is one "step", right?

For other examples, you can look at the difference between variant 2&3 https://github.com/The-OpenROAD-Project/megaboom/pull/176.

3_4_place_resized.log, 3_5_place_dp.log, 4_1_cts.log are all exhibiting a 3x+ longer time in report_metrics than running time for the steps.