The-OpenROAD-Project/OpenROAD #6033: Support partial automatic macro placement

oharboe, 2024-10-28:
### Description

untar  [partial-automatic-macro-placement.tar.gz](https://github.com/user-attachments/files/17539427/partial-automatic-macro-placement.tar.gz)

### Suggested Solution

Support partial automatic macro placement

### Additional Context

If support is added, I'll update https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2516

AcKoucher, 2024-11-12:
@oharboe I had a discussion with @maliberty  and, before actually digging into the implementation of partial placement, our idea is to allow the already existing mpl2 guidance mechanism to be used. We support it internally, but there's no .tcl API to actually used it.

Basically we'd let the user specify a certain region in which the macro must be placed in order to guide the automated placement. I think that can help solving the issue.

Having a mechanism capable of incorporating the data of a placed macro into the floor-plan packing during SA would be challenging to implement. It'd be a bigger project.

oharboe, 2024-11-13:
> @oharboe I had a discussion with @maliberty  and, before actually digging into the implementation of partial placement, our idea is to allow the already existing mpl2 guidance mechanism to be used. [...]

Specifying regions sounds like a better feature. For megaboom, I could specify that integer register file with pins on horizontal edge is placed near the center top of the design and that the floating point register file with pins on vertical edge is placed near middle of the left edge, for instance.

However, how small can these regions be?  A single location? What should the user expect if he chooses a region that is effectively a single location? Just curious if this is supported and how it is handled.

AcKoucher, 2024-11-13:
> However, how small can these regions be? A single location? What should the user expect if he chooses a region that is effectively a single location? Just curious if this is supported and how it is handled.

We'll try to align the center of the macro with the center of the specified region. With regards to how small it could be, I don't see a reason to have a lower limit. However, I think that using a single location would be less practical - and a bit odd - than just specifying a rectangle that covers a somewhat interesting region for the macro. That's just my opinion though.

Note: I'm interpreting a single location as a region made of a single point:
( x1 , y1 ), ( x2 , y2 ) 
where x1 = x2 and y1 = y2.

AcKoucher, 2024-11-14:
> BTW what happens if I want to have multiple macros inside a rectangle? (a bank of sram macros).

You could set the same region for all these macros (in this case the region would cover the area of all the macros).
However, there's no guarantee that annealing will find the perfect desired position. The macros will be guided to that area during the convergence process.

I think it will be clearer when testing.

AcKoucher, 2024-11-19:
Created #6200 for the guidance regions. I finished the first initial version and found bugs in the original code when testing. @joaomai is taking over the debugging part.