The-OpenROAD-Project/OpenROAD #6034: CTS WNS drops in timing repair before ending up roughly at where it started

oharboe, 2024-10-28:
### Describe the bug

Here WNS gets wors in timing repair in CTS before bouncing back up to roughly where it started.

[image]

untar https://drive.google.com/file/d/1ICisYfJdyxyJbMKcgs6JkNTB2IT05afA/view?usp=sharing

```
./run-me-BoomTile-asap7-4.sh
```

### Expected Behavior

I don't know what the expected behavior is here, but I tought it looked interesting enough to file a standalone test-case.

### Environment

```shell
OpenROAD v2.0-16787-gcd519bb5e
```

### To Reproduce

See above

maliberty, 2024-10-28:
We do allow some degradation in order to avoid local minima but this seems a bit extreme.

precisionmoon, 2024-10-28:
Can you clarify the issue a bit?  Are you concerned about setup WNS degradation during hold fixing?  Or is this about setup WNS QoR trajectory from floorplanning to CTS?

oharboe, 2024-10-28:
I dont have any concerns or expectations as I dont know the repair setup algorithm, I just thought the WNS dip during setup fixing might be an interesting test case that merited some further study: do we understand what is happening here?

precisionmoon, 2024-10-28:
OK.  Understood.   We'll look into WNS fluctuation during setup fixing.

precisionmoon, 2024-10-28:
I'm getting an incompatible db schema error after re-building OR with the latest version:
Error: cts.tcl, 4 incompatible database schema revision 0.96 > 0.90

maliberty, 2024-10-28:
That would suggest you have an older version rather than a newer one.  What commit id does OR show?

precisionmoon, 2024-10-28:
OpenROAD v2.0-15833-ge3b4fc0c9

precisionmoon, 2024-10-28:
Sorry, I had an issue with git pull which caused the db schema error.  Now I can reproduce the issue.

oharboe, 2024-10-30:
Here is another plot from various runs, it looks like the test case is indeed a bit peculiar and merits some further study:

[image]