The-OpenROAD-Project/OpenROAD #6037: [WARNING GRT-0026] Missing route to pin _766_/ZN.

oharboe, 2024-10-28:
### Describe the bug

untar [warnings.tar.gz](https://github.com/user-attachments/files/17542319/warnings.tar.gz)

```
$ ./run-me-gcd-nangate45-base.sh 
OpenROAD v2.0-16787-gcd519bb5e 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
GUI_TIMING=1 reading timing, takes a little while for large designs...
estimate_parasitics -global_routing
[WARNING GRT-0026] Missing route to pin _766_/ZN.
[WARNING GRT-0026] Missing route to pin dpath.a_reg.out\[1\]$_DFFE_PP_/D.
[WARNING GRT-0026] Missing route to pin _815_/ZN.
[WARNING GRT-0026] Missing route to pin _829_/ZN.
[deleted]
Populating timing paths...
openroad>
```

Can also be reproduced by default target designs/nangate45/gcd/config.mk:

```
$ make nuke
$ make route
$ make open_route
$ make open_route
ODB_FILE=./results/nangate45/gcd/base/5_route.odb /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -no_init -threads 48  /home/oyvind/OpenROAD-flow-scripts/flow/scripts/open.tcl
OpenROAD v2.0-16787-gcd519bb5e 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 48 thread(s).
GUI_TIMING=1 reading timing, takes a little while for large designs...
estimate_parasitics -global_routing
[WARNING GRT-0026] Missing route to pin _766_/ZN.
[WARNING GRT-0026] Missing route to pin dpath.a_reg.out\[1\]$_DFFE_PP_/D.
[WARNING GRT-0026] Missing route to pin _815_/ZN.
[WARNING GRT-0026] Missing route to pin _829_/ZN.
[WARNING GRT-0026] Missing route to pin dpath.b_reg.out\[0\]$_DFFE_PP_/D.
[WARNING GRT-0026] Missing route to pin _893_/ZN.
[WARNING GRT-0026] Missing route to pin dpath.b_reg.out\[12\]$_DFFE_PP_/D.
[WARNING GRT-0026] Missing route to pin _906_/ZN.
[WARNING GRT-0026] Missing route to pin dpath.b_reg.out\[14\]$_DFFE_PP_/D.
[deleted]
Populating timing paths...
openroad> 
```

### Expected Behavior

No warnings

### Environment

```shell
OpenROAD v2.0-16787-gcd519bb5e
```

### To Reproduce

See above

eder-matheus, 2024-11-07:
@oharboe I created two PRs that will solve the warnings that you're facing, but I'm curious: why are you estimating the parasitics from global routing when you have the detailed routing of the design? Couldn't you just extract the actual parasitics with rcx?

oharboe, 2024-11-07:
> @oharboe I created two PRs that will solve the warnings that you're facing, but I'm curious: why are you estimating the parasitics from global routing when you have the detailed routing of the design? Couldn't you just extract the actual parasitics with rcx?

I am using open.tcl in automation to read out WNS at global route for instance. Or I am looking at the global route results using open.tcl (make gui_open).

eder-matheus, 2024-11-07:
> If I included the detailed route in the reproducible example, they may have been from a previous run.

Actually, when I run `make route` and then `make open_route`, it load the odb file with the detailed routing on it. That's why I was confuse. But in any case, GRT should not produce this warnings, and the fixes are on the way.

Thanks for the clarifications.

maliberty, 2024-11-07:
You want open_grt rather than open_route.

closed 2024-11-08 by maliberty