The-OpenROAD-Project/OpenROAD #6038: PDN missing power and ground connections to IO

lanserge, 2024-10-28:
### Describe the bug

add_pdn_ring command with -connect_to_pads parameter produce no power and ground connections to IO for IHP sg13g2. Same command properly works for gf130 but not for IHP sg13g2.

### Expected Behavior

Power and ground should be connected from created PDN ring  to IO.

### Environment

```shell
kernel: Linux 6.10.6+bpo-amd64
os: Debian GNU/Linux 12 (bookworm)
cmake version 3.30.5
-- The CXX compiler identification is GNU 12.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16796-g2bb84c2ea
-- System name: Linux
-- Compiler: GNU 12.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare - Success
[... 87 lines cut ...]
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization system thread
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")
-- Found Eigen3: /home/serge/projects/chipflow-backend-tools/tools/or-tools/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
```

### To Reproduce

Unpack attached zip
[2_6_floorplan_pdn.zip](https://github.com/user-attachments/files/17545542/2_6_floorplan_pdn.zip)
 file (includes all dependencies), from unpacked directory run `make`.
The result 2_6_floorplan_pdn.odb will be in the results sub-directory.
`make pnr_2_6_floorplan_pdn_images` will make also images from ODB files.
`make pnr_2_6_floorplan_pdn_verilog` will produce netlist verilog files.
`make all` will produce everything above.
`make clean` will remove all produced files (to start over).

### Screenshots

[image]

gadfort, 2024-10-28:
@lanserge due to the way the pads are constructed, the connections for to the rings have to come from above. Since that is the case you need to define the connection rules that will support that.

```
add_pdn_connect -grid {grid} -layers {Metal5 TopMetal2}
add_pdn_connect -grid {grid} -layers {TopMetal1 TopMetal2}
```
With those added I get:
[image]

maliberty, 2024-10-28:
close?

gadfort, 2024-10-28:
@maliberty I think it can be closed.

closed 2024-10-29 by maliberty

lanserge, 2024-10-29:
It works, thank you very much!

It actually needs only `add_pdn_connect -grid {grid} -layers {TopMetal1 TopMetal2}` as `add_pdn_connect -grid {grid} -layers {Metal5 TopMetal1}` exist and together they will make `{Metal5 TopMetal2}` connection.
Didn't realize that PDN needs full propagation from the pads themselves (TopMetal2 is only in pads).

Thank you for such swift response with working solution.